alexa-pi/AlexaPiDEPRECATED #82: Active Listening.  Maybe the saving grace?

gregwa1953, 2016-05-13:
Saw this on the Java Alexa version, so I thought that I would share it here.
Sorry if someone already did this.

Gregwa

GemBro, 2016-05-15:
**@gregwa1953**

You got the link Greg? ... I couldn't find it ... cheers ...

gregwa1953, 2016-05-15:
Sorry about that @GemBro.

The main issue page is:
https://github.com/amzn/alexa-avs-raspberry-pi/issues/4#issuecomment-218787833

Here's the text from the post...

xuchen <https://github.com/xuchen> commented 3 days ago
<https://github.com/amzn/alexa-avs-raspberry-pi/issues/4#issuecomment-218787833>

Sorry for coming late into this discussion. We just released a hotword (or
"wake word" as Amazon calls it) detector specifically for voice triggering
an action:

https://snowboy.kitt.ai

It has Raspberry Pi support and Python bindings/wrapper (through SWIG).
Though no Java wrapper yet:

https://github.com/kitt-ai/snowboy

We'd very much appreciate a pull request for Alexa/Pi integration.

flooie, 2016-05-15:
I couldn't make it - anyone else try?

gregwa1953, 2016-05-16:
Haven't had a chance to yet.
Maybe tomorrow.

GemBro, 2016-05-16:
**@gregwa1953**

That seems a bit of a long winded way :) ... I'm not playing with the AVS (Amazon) version at the moment, as I prefer a headless system ... but I love all the ideas that are coming out of this project ...

gregwa1953, 2016-05-16:
Well, with a bit of code, it COULD replace the button.  You would have to
change the button input to a jump into the python program and change the
need for pressing the button until you are through with your vocal input to
... say ... 10 second sampling.

Not saying it's the ultimate answer to get rid of the button, but it might
be fun to try.

flooie, 2016-05-16:
I've had trouble making the snowboy project on my rpi - anyone else have any luck?

flooie, 2016-05-16:
I got it working- and its awesome.

maso27, 2016-05-16:
I haven't tried snowboy, but I've had voice control working via CMU Sphinx for a few weeks now in my office.  

I grabbed a number of things from [https://github.com/fcooper/beagleman](url) to get it running.

I wedged it into Sam's latest version1.1 this morning, and if you want you can try it here:
[https://github.com/maso27/AlexaPi/tree/version1.1](url)

tube0013, 2016-05-16:
That's awesome, can you share the code?

flooie, 2016-05-16:
@tube0013 planning on it - just working on one kink - right now i can't command it to stop the previous command -

flooie, 2016-05-16:
@tube0013 anyone know how to detect a file executing

GemBro, 2016-05-16:
Nice one flooie ... so I take it the 'button' does not stop the previous command? ...

flooie, 2016-05-16:
@tube0013 - i posted my code - - I haven't figured out how to stop Alexa if its playing with voice commands but it works and works well.  you are going to need to add a few more modules - and i had trouble getting the microphone code up to snuff but feel free to ping me if you have trouble

tube0013, 2016-05-17:
My pi is at my office, and I was home today, so I'll give it a go tomorrow.  I just saw that the snowbird folks added in silence detection:

https://github.com/Kitt-AI/snowboy/issues/3

GemBro, 2016-05-17:
Blimey ... this has taken off a bit eh? ... great work ... good to see it's all coming together nicely ... now to keep up with all the additions being made ... :) ...

**@flooie**

Are you and Sam talking? ... hope your heads are getting together ...

benwoodley, 2016-05-17:
In that issue I gave a mock up solution btw, but it isn't amazing for long term.

GemBro, 2016-05-17:
**@benwoodley** 
Yes, sorry Ben, I saw that ... have ammended the above comment ;) ...

flooie, 2016-05-17:
@GemBro @tube0013 I have everything working (albeit a bit slow) - and I posted a short video of it running in my kitchen - switching accounts playing music etc. etc.

https://youtu.be/wLbsAQDmN-c

gregwa1953, 2016-05-17:
Thanks to everyone for all their hard work in integrating this into the
project.  I'm glad that I was able to help out by pointing this out.

Gregwa

GemBro, 2016-05-17:
**@flooie** 

It maybe slow but that is totally awesome mate ... very well done ... love the feedback tones ... I see you got Alexa to actually stop now too ... nice ... I understand you're a bit of a distance away from the unit hence the couple of tries ... :) ...

One thing flooie ... I tried to install your VC version last night and I couldn't get the setup.sh to run ... v1.1 runs fine but I may have it all wrong when cloning your GIT files ...

Will you be doing a simple step by step installation write up for this? ... I find installing GIT stuff a bit hard, as I'm new to it all ... :) ...

**@greg**
Yes indeed ... good call mate ... thanks for pointing us all in that direction ... ;) ...

tube0013, 2016-05-17:
@flooie I pulled this down, but I'm having issues with the pyaudio throwing an error at start, did you have to do anything to get this part to work with snowdecoder.py?

flooie, 2016-05-17:
@tube0013 - yeah - i had a heck of time getting pyaudio to work.  

Look at the stack questions below - they should probably help you figure out what going on. 
Before you try to run my code you need to test that you can execute pyaudio - I ended up having to adjust a couple files.  - also you should have everything installed that snowboy requires.  

Hope that helps

http://raspberrypi.stackexchange.com/questions/9951/pyaudio-recording-sound-on-pi-getting-errors

http://raspberrypi.stackexchange.com/questions/39928/unable-to-set-default-input-and-output-audio-device-on-raspberry-jessie

https://github.com/raspberrypi/linux/issues/994

xuchen, 2016-05-17:
Coming late into this conversation: glad that some of you got it working! We actually wrote in the documentation how to configure pyaudio on the Pi:

https://snowboy.kitt.ai/docs/#set-up-audio

Hope it helps!

xuchen, 2016-05-17:
Two more tips, if you want to have it working across the room better you can do two things:

1. adjust sensitivity level
2. when training the model, record your voice from far in the room. You can also record it on your Pi with your microphone, and upload the .wav files to snowboy website to train a new model that fits exactly to your microphone and room settings.

tube0013, 2016-05-17:
I've set my default cards up in the .asound and also in /etc/asoound.conf. 

I've also tried changing the default card to my usb mic (which also has audio out) in the /usr/share/alsa/alsa.conf file.  no matter what I get errors like this:

```
$ python demo.py Alexa.pmdl 
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, self->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2719
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
Traceback (most recent call last):
  File "demo.py", line 27, in <module>
    detector = snowboydecoder.HotwordDetector(model, sensitivity=0.5)
  File "/home/pi/Downloads/rpi-arm-raspbian-8.0-1.0.0/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "build/bdist.linux-armv7l/egg/pyaudio.py", line 750, in open
    
  File "build/bdist.linux-armv7l/egg/pyaudio.py", line 441, in __init__
    # calling pa.open returns a stream object
IOError: [Errno -9997] Invalid sample rate
```

flooie, 2016-05-17:
@tube0013  http://superuser.com/questions/989385/alsa-base-conf-missing-in-new-raspberry-pi-raspbian-jesse

Found it - look at the answer from  Digitalfix 

I bet that will fix it

tube0013, 2016-05-17:
@flooie thanks. just tried that, and it's a no go.  not sure but may have to do with sample rate and chunk size based on some other posts I've seen.  I don't get the IOError when I run start.py from your files, I just get a Traceback.  the demo from snowboy gives the Sample Rate error.

xuchen, 2016-05-17:
@tube0013 did you follow our docs: https://snowboy.kitt.ai/docs/#set-up-audio

tube0013, 2016-05-17:
Yes, that's where I started from.

xuchen, 2016-05-17:
@tube0013 i'd try to uninstall PulseAudio and only use ALSA.

tube0013, 2016-05-17:
I'm going to start from a fresh Rasbian/Jessie install tomorrow.  I thought I had pulse completely uninstalled.

chenguoguo, 2016-05-18:
In addition to what Xuchen said regarding using Snowboy  with a distance, I'd add one more trick: increasing the audio gain in the snowboy API.

xuchen, 2016-05-18:
@flooie any idea why it's a bit slow to respond, esp. when you turn on BBC?

flooie, 2016-05-18:
1) I haven't implemented silence listening so it listens for a set period of time- 
2) My code collects all of the content before it starts playing it so its slightly delayed there 
3) *The internet sucks in Cambridge, MA.

tube0013, 2016-05-18:
I'm still having the same issues after a complete re-flash of the latest raspbian.

I managed to get the demo.py provided in the snowbird download working.  Weirdly it only works if I run as root/sudo.

for voice-command AlexaPi, I'm running python start.py Alexa.pmdl,

and I get an error:
no Module named subfile

is this a prereq I'm missing?  if so can you point me to it?

flooie, 2016-05-18:
@tube0013 email me wpalin [at] gmail.com

GemBro, 2016-05-19:
**@tube0013**

Have you tried running the files using sudo? ... like this:

sudo python start.py Alexa.pmdl

I have to use sudo sometimes if it cannot find files that are there and visible ...

tube0013, 2016-05-19:
Yes I got it running. Only works as root/sudo.

Had to delete the import subfile line, @flooie made the change in his
repository today.

GemBro, 2016-05-19:
**@tube0013**

Cool ... thanks for heads up ... gonna grab the new update ...

olsonk, 2016-05-24:
Hey Everyone,

Really appreciate the work that's gone into this so already! I tried installing everything, though, and running with `python start.py Alexa.pmdl` and am getting this error:

```
pi@raspberrypi:~/Downloads/AlexaPi $ python start.py Alexa.pmdl
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Listening... Press Ctrl+C to exit
INFO:snowboy:Keyword 1 detected at time: 2016-05-24 09:31:22
Ready to Record.
Recording...
[... cut ...]
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 747, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 442, in __init__
    self._stream = pa.open(**arguments)
IOError: [Errno Device unavailable] -9985
```
Any ideas what might be going on there? I tried running the Snowboy demo and it worked properly, and I can use `rec temp.wav` and `play temp.wav` to record and playback audio properly. I'm not sure exactly what's going on...

tube0013, 2016-05-24:
I had to run it as root or I'd get similar errors. I also had to make sure my default audio devices were set up.  My Microphone also has an audio output, so I was unable to pick just the mic as default and had to plug my speaker into the usb output. and change the default device in /usr/share/asla/alsa.conf

olsonk, 2016-05-24:
@tube0013 Thanks for responding. I'm actually getting errors that prevent it from even listening when I run as root! Here's the printout from running `sudo python start.py Alexa.pmdl`:

```
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Connection refused

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
[... cut ...]
I thought I'd properly selected my audio devices by editing the `~/.asoundrc` file, which is referenced in  `/usr/share/alsa/alsa.conf`. Do I also need to change other settings in `alsa.conf`?

I'm pretty new to all of this stuff, so I apologize if I'm asking stupid questions or missed something obvious...

Here's my `~/.asoundrc` file just in case it's helpful:

```
pcm.!default {
  type asym
   playback.pcm {
     type plug
     slave.pcm "hw:0,0"
   }
   capture.pcm {
     type plug
     slave.pcm "hw:1,0"
   }
}
```

Thanks for your help!

chenguoguo, 2016-05-24:
@sciencedude100 is having similar problems, see the issue here: https://github.com/Kitt-AI/snowboy/issues/9

I posted a potential solution there. The idea is to manually set device for PyAudio, not sure if that's going to work. @olsonk, it will be great if you can give it a try.

tube0013, 2016-05-24:
@olsonk I was getting that same error.  I ended up re installing Jessie and starting from scratch.  And still had to update /usr/share/alsa/alsa.conf.  @chenguoguo may be on to something by manually setting the device for PyAudio.

olsonk, 2016-05-24:
Ha - I literally just googled "pyaudio set input device" after thinking more about the issue - seems as though PyAudio is setting off the IOError. 

So, I followed your solution on that snowboy issue, @chenguoguo . I edited self.steam_in to reflect your code and found that my USB mic is index 2. I added `input_device_index=2` and now I get this error:

```
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, self->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2719
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
Traceback (most recent call last):
  File "start.py", line 32, in <module>
    detector = snowboydecoder.HotwordDetector(model, sensitivity=0.5)
  File "/home/pi/Downloads/AlexaPi/snowboydecoder.py", line 123, in __init__
    input_device_index=2)
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 747, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pyaudio.py", line 442, in __init__
    self._stream = pa.open(**arguments)
IOError: [Errno Invalid sample rate] -9997
```

All of that is after the other errors and warnings from above. Here's snowboydecoder.py, the relevant section:
```
self.audio = pyaudio.PyAudio()
        self.stream_in = self.audio.open(
            input=True, output=False,
            format=pyaudio.paInt16,
            channels=1,
            rate=16000,
            frames_per_buffer=2048,
            stream_callback=audio_callback,
            input_device_index=2)
```

Anything jump out to anyone?

olsonk, 2016-05-24:
I just adjusted the rate to 48000 and am no longer getting the Invalid Sample Rate error. However, it also won't detect the keyword! Is there any way for me to figure out the proper sample rate for the USB mic? Or is that even important? Also wondering if there's some way I can test whether it's picking up any audio from the mic with that stream and maybe I just need to speak differently?

chenguoguo, 2016-05-24:
@olsonk, so looks like you USB micphone only supports a sampling rate of 48000? It's expected that keyword spotting stopped working with that sampling rate, as Snowboy requires a sampling rate of 16000 which is pretty standard for speech recognition tasks...

flooie, 2016-05-24:
@olsonk which branch are you trying to run with? -

xuchen, 2016-05-24:
@olsonk did you try your own .pmdl or someone else's .pmdl? You can also try the universal model for the word "snowboy":

https://github.com/Kitt-AI/snowboy/raw/master/resources/snowboy.umdl

olsonk, 2016-05-24:
Sorry, was away for a bit and will be away again shortly...

@chenguoguo yeah I don't know what's going on with the sampling rate. I've tried two different USB mics, one of which is fancy-looking, and they both said they support 44100 as a sampling rate but not 16000.

@flooie I downloaded from your voice-command branch

@xuchen yes, I tried my own Alexa.pmdl from snowboy, and also have tested with the universal model you included.

I'm wondering if there's just a configuration option that got messed up somewhere along the way. I followed the https://github.com/amzn/alexa-avs-raspberry-pi project initially, then shifted to this one. Maybe something in those two projects is conflicting? The Amazon project is not running on the Pi, but it did require that I remove VLC and install a bunch of other packages.

I think I'll try with a totally fresh Raspbian and see if that takes care of things. Thanks so much for your help, though!

flooie, 2016-05-24:
@olsonk - the one on my fork isn't universal - its my personal model (pmdl)  Im going to remove it so people have to make their own.  I believe that is what snowboy would prefer.

olsonk, 2016-05-24:
Is this helpful at all? I ran system_info.py from PyAudio's git and got these results for my devices. The USB mic in question is Index 2:
```
Devices:
========
	defaultSampleRate: 44100.0
	defaultLowOutputLatency: 0.00580498866213
	defaultLowInputLatency: -1.0
	maxInputChannels: 0
	structVersion: 2
	hostApi: 0 (ALSA)
	index: 0
	defaultHighOutputLatency: 0.0348299319728
	maxOutputChannels: 2
	name: bcm2835 ALSA: - (hw:0,0)
	defaultHighInputLatency: -1.0
	Output rates: [8000.0, 9600.0, 11025.0, 12000.0, 16000.0, 22050.0, 24000.0, 32000.0, 44100.0, 48000.0]
	--------------------------------
	defaultSampleRate: 44100.0
	defaultLowOutputLatency: 0.00580498866213
	defaultLowInputLatency: -1.0
	maxInputChannels: 0
	structVersion: 2
	hostApi: 0 (ALSA)
	index: 1
	defaultHighOutputLatency: 0.0348299319728
	maxOutputChannels: 2
	name: bcm2835 ALSA: IEC958/HDMI (hw:0,1)
	defaultHighInputLatency: -1.0
	Output rates: [44100.0, 48000.0]
	--------------------------------
	defaultSampleRate: 44100.0
	defaultLowOutputLatency: -1.0
	defaultLowInputLatency: 0.00868480725624
	maxInputChannels: 1
	structVersion: 2
	hostApi: 0 (ALSA)
	index: 2
	defaultHighOutputLatency: -1.0
	maxOutputChannels: 0
	name: AK5370: USB Audio (hw:1,0)
	defaultHighInputLatency: 0.0348299319728
	Input rates: [8000.0, 11025.0, 22050.0, 44100.0, 48000.0]
	--------------------------------
	defaultSampleRate: 44100.0
	defaultLowOutputLatency: 0.00580498866213
	defaultLowInputLatency: -1.0
	maxInputChannels: 0
	structVersion: 2
	hostApi: 0 (ALSA)
	index: 3
	defaultHighOutputLatency: 0.0348299319728
	maxOutputChannels: 128
	name: sysdefault
	defaultHighInputLatency: -1.0
	Output rates: [8000.0, 9600.0, 11025.0, 12000.0, 16000.0, 22050.0, 24000.0, 32000.0, 44100.0, 48000.0, 88200.0]
	--------------------------------
	defaultSampleRate: 48000.0
	defaultLowOutputLatency: 0.0213333333333
	defaultLowInputLatency: -1.0
	maxInputChannels: 0
	structVersion: 2
	hostApi: 0 (ALSA)
	index: 4
	defaultHighOutputLatency: 0.0213333333333
	maxOutputChannels: 2
	name: dmix
	defaultHighInputLatency: -1.0
	Output rates: [48000.0]
	--------------------------------
	defaultSampleRate: 44100.0
	defaultLowOutputLatency: 0.00580498866213
	defaultLowInputLatency: 0.00868480725624
	maxInputChannels: 128
	structVersion: 2
	hostApi: 0 (ALSA)
	index: 5
	defaultHighOutputLatency: 0.0348299319728
	maxOutputChannels: 128
	name: default
	defaultHighInputLatency: 0.0348299319728
	Input rates: [8000.0, 9600.0, 11025.0, 22050.0, 32000.0, 44100.0, 48000.0, 88200.0]
	Output rates: [8000.0, 9600.0, 11025.0, 12000.0, 16000.0, 22050.0, 24000.0, 32000.0, 44100.0, 48000.0, 88200.0]
	Full duplex: [8000.0, 9600.0, 11025.0, 22050.0, 32000.0, 44100.0, 48000.0, 88200.0]
	--------------------------------

Default Devices:
================
Default Input Device : 5
	defaultSampleRate: 44100.0
	defaultLowOutputLatency: 0.00580498866213
	defaultLowInputLatency: 0.00868480725624
	maxInputChannels: 128
	structVersion: 2
	hostApi: 0 (ALSA)
	index: 5
	defaultHighOutputLatency: 0.0348299319728
	maxOutputChannels: 128
	name: default
	defaultHighInputLatency: 0.0348299319728
	--------------------------------
Default Output Device: 5
	defaultSampleRate: 44100.0
	defaultLowOutputLatency: 0.00580498866213
	defaultLowInputLatency: 0.00868480725624
	maxInputChannels: 128
	structVersion: 2
	hostApi: 0 (ALSA)
	index: 5
	defaultHighOutputLatency: 0.0348299319728
	maxOutputChannels: 128
	name: default
	defaultHighInputLatency: 0.0348299319728
```
I tried using 11025 for the sampling rate, but still no luck. Will keep plugging away and see if I can figure this out...

flooie, 2016-05-24:
@olsonk - and just to confirm - you are able to record your voice using pulse audio. something like this?

import pyaudio
import wave

def recordAudio():

	FORMAT = pyaudio.paInt16
	CHANNELS = 1
	RATE = 16000
	CHUNK = 500
	RECORD_SECONDS = 5
	WAVE_OUTPUT_FILENAME = "recording.wav"

	audio = pyaudio.PyAudio()
	# play_audio_file(DETECT_DONG)
	# start Recording
	stream = audio.open(format=FORMAT,
						channels=CHANNELS,
						rate=RATE,
						input=True,
						frames_per_buffer=CHUNK)
	print "recording..."
	frames = []

	for i in range(0, int(RATE / CHUNK * RECORD_SECONDS)):
		data = stream.read(CHUNK)
		frames.append(data)
	print "finished recording"

	# stop Recording
	stream.stop_stream()
	stream.close()
	audio.terminate()

	waveFile = wave.open(WAVE_OUTPUT_FILENAME, 'wb')
	waveFile.setnchannels(CHANNELS)
	waveFile.setsampwidth(audio.get_sample_size(FORMAT))
	waveFile.setframerate(RATE)
	waveFile.writeframes(b''.join(frames))
	waveFile.close()
	# play_audio_file(DETECT_DING)

recordAudio()

olsonk, 2016-05-25:
Alright, so I am now trying this on a fresh Raspbian Jessie install. I seem to have things working properly in terms of the code (I can start start.py, it listens for the Alexa hotword and responds with "recording...", and gets a response back from Alexa). My issue now is that I can't hear anything from my pi!

Anytime I enter `amixer cset numid=3 1` to change to headphones, the volume control goes to 0 (mute) and I hear nothing, even when I drag the slider up to 100 again. I can't get any sound output from Sonic Pi either. Here's my `aplay -l` output:

```
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

and `arecord -l` output:
```
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

and `~/.asoundrc` file:
```
pcm.!default {
  type asym
   playback.pcm {
      type plug
      slave.pcm "hw:0,0"
   }
   capture.pcm {
      type plug
      slave.pcm "hw:1,0"
   }
}
```

Anything look odd? There must be some issue with ALSA config or something that's not outputting sound to the proper place. When I listen to recordings from my mic on another computer via VNC, the recording is properly capturing my voice. It's just sound output now that's got me tripped up...

xuchen, 2016-05-25:
Is your HDMI connected with a monitor and Pi by default chooses to output
audio through HDMI?

This might help:

https://www.raspberrypi.org/documentation/configuration/audio-config.md

olsonk, 2016-05-25:
@flooie On my fresh install, I have your PyAudio test working (recordAudio()) and it captures a recording properly.

I uninstalled PulseAudio and things seem to be working better; I can capture my voice, run the Snowboy demo properly, and hear my audio output. Now, when I run `python start.py Alexa.pmdl`, I get the following error after it detects the hotword for a few seconds:
```
Listening... Press Ctrl+C to exit
INFO:snowboy:Keyword 1 detected at time: 2016-05-25 19:26:04
Ready to Record.
Recording...
Expression 'ret' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1736
Expression 'AlsaOpen( &alsaApi->baseHostApiRep, params, streamDir, &self->pcm )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1904
Expression 'PaAlsaStreamComponent_Initialize( &self->capture, alsaApi, inParams, StreamDirection_In, NULL != callback )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2171
Expression 'PaAlsaStream_Initialize( stream, alsaHostApi, inputParameters, outputParameters, sampleRate, framesPerBuffer, callback, streamFlags, userData )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2840
Traceback (most recent call last):
  File "start.py", line 37, in <module>
    sleep_time=0.03)
  File "/home/pi/Downloads/AlexaPi/snowboydecoder.py", line 187, in start
    main.start()
  File "/home/pi/Downloads/AlexaPi/main.py", line 61, in start
    recordAudio()
  File "/home/pi/Downloads/AlexaPi/main.py", line 82, in recordAudio
    frames_per_buffer=CHUNK)
  File "build/bdist.linux-armv7l/egg/pyaudio.py", line 750, in open
    
  File "build/bdist.linux-armv7l/egg/pyaudio.py", line 441, in __init__
    # calling pa.open returns a stream object
IOError: [Errno -9985] Device unavailable
```
Looks like an issue in my pa_linux_alsa.c file? Not sure if that 'pa' is PulseAudio or PyAudio, or what the difference even is between those two...

olsonk, 2016-05-25:
I've fiddled a bit with `snowboydecoder.py` and think maybe I've found something. I believe my input device is unable to record my Alexa command stream because of the Snowboy stream that's already running? I added `self.stream_in.stop_stream()` and `self.stream_in.close()` right before `main.setup()` on line 189-ish of `snowboydecoder.py` and I was able to actually record a message for Alexa! However, afterward I got some JSON back (not sure if it has private information tucked within) and this:
```
 ------->>>>> audio item
 
download content and play first file
0
>>----------------------------------> SPEAK
find any streams
add anything else
>>-------------------------->>> Audio Items
[]
check for next round
------------------------>>> should we continue?
----------------------- >>>>>>>> end
Traceback (most recent call last):
>>--->>  CALLBACK <<---<<
  File "start.py", line 35, in <module>
State.Opening
    set player
detector.start(detected_callback=snowboydecoder.play_audio_file, interrupt_check=interrupt_callback, sleep_time=0.03)
  File "/home/pi/Downloads/AlexaPi/snowboydecoder.py", line 159, in start
    self.stream_in.start_stream()
  File "build/bdist.linux-armv7l/egg/pyaudio.py", line 511, in start_stream
    
IOError: [Errno -9988] Stream closed
```
Which doesn't surprise me, since I closed the stream in my changes! I also didn't get a spoken response back from Alexa...

What I can't figure out now is how to set it up so that a stream from Snowboy starts, monitors for the hotword, closes the stream when the hotword is detected, uses `main.py` to capture the next few seconds of speech, sends that over to Alexa, and then re-starts the Snowboy detector for my next command.

Or am I totally off-base? I must admit, I'm not a professional programmer (could you tell? ;-) ) and may just be missing something totally obvious.

...Help?

hossame, 2016-05-27:
Hi oslonk,

I was experiencing your exact same error, Errno -9988.

I then ran the following command and I've got everything working:

`pulseaudio -D`

korniza, 2016-05-29:
Hi guys!
I have the same issue 

`IOError: [Errno Device unavailable] -9985`

everything else is working fine. I tried to use @olsonk workaround. It worked for the first prompt but after it did not respond to snowboy. I used @flooie voice-command repo. Hope you can find a solution.

[61 more replies]

closed 2016-10-26 by renekliment