alexa-pi/AlexaPiDEPRECATED #83: Correct GPIO pinout

Tiqni, 2016-05-14:
"A push to make button connected between GPIO 18 and GND"
This made me a headache, I couldn't understand why my setup is not working...

After trying each GPIO, I understood that it means "BCM 18" which is GPIO 12 pin.
(I'm using Raspberry Pi 3)

benwoodley, 2016-05-17:
If you need it to use a different pin then just change it in the main.py file, the pin is GPIO 18. GPIO 18 is pin 12 on the Pi 3. I don't think there is any difference between the pins on pi 2 and 3. The Pi 3 has very little design differences, just speed and wifi ect.
Use the diagram below for reference.
[image]

sammachin, 2016-05-19:
The Naming is pretty clear IMO  and the project also contains a digram showing the wiring, closing.

closed 2016-05-19 by sammachin