alexa-pi/AlexaPiDEPRECATED #86: Bluetooth speaker integration

glenndav, 2016-05-17:
While audio through the wired output works fine in this project, and while my Bluetooth speaker works well with other applications on my Pi, there appears to be a conflict which prevents Bluetooth audio out here - even when the Bluetooth speaker is the default audio output.

flooie, 2016-05-17:
@glenndav I use a bluetooth speaker - which branch are you using?

glenndav, 2016-05-17:
1.1 branch

flooie, 2016-05-17:
Back trace?

glenndav, 2016-05-17:
I think it is a bad configuration of bluetooth on my end... although LXMusic and VLC work fine with my bluetooth speakers through the GUI, it is not showing up when I type:

> arecord -L

`default
    Playback/recording through the PulseAudio sound server
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
sysdefault:CARD=TP6920
    TP6920, USB Audio
    Default Audio Device
front:CARD=TP6920,DEV=0
    TP6920, USB Audio
    Front speakers
surround21:CARD=TP6920,DEV=0
    TP6920, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=TP6920,DEV=0
    TP6920, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=TP6920,DEV=0
    TP6920, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=TP6920,DEV=0
    TP6920, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=TP6920,DEV=0
    TP6920, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=TP6920,DEV=0
    TP6920, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=TP6920,DEV=0
    TP6920, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=TP6920,DEV=0
    TP6920, USB Audio
    Direct sample mixing device
dsnoop:CARD=TP6920,DEV=0
    TP6920, USB Audio
    Direct sample snooping device
hw:CARD=TP6920,DEV=0
    TP6920, USB Audio
    Direct hardware device without any conversions
plughw:CARD=TP6920,DEV=0
    TP6920, USB Audio
    Hardware device with all software conversions`

flooie, 2016-05-17:
@glenndav I would plug into a screen and open up pulse and see if the audio is still trying to be played thru something else. and build a unit test to play the hello.mp3 until you get it working.

glenndav, 2016-05-18:
Thanks, a lot.

closed 2016-05-18 by glenndav