alexa-pi/AlexaPiDEPRECATED #87: Internet connection issues

twitherington, 2016-05-19:
The Pi boots up and I do hear a "Hello" from Alexa, however I don't get any further. 
Looking at the /var/log/alexa.log I see the following.

```
Starting Alexa...
Checking Internet Connection
Connection Failed
.
Checking Internet Connection
Connection Failed
.
```
and continuing like so. 

I have both eth0 and wlan0 are setup and I'm getting a valid IP for both. Pulling up a browser I do have an internet connection, so I'm unsure what to do at this point.

sammachin, 2016-05-21:
@twitherington Is that all you see in the log? Usually you will get a couple of connection failed then a connection ok? 

Which branch are you using?

twitherington, 2016-05-21:
Master branch

sammachin, 2016-05-22:
MIght be a silly question but you are holding down the button for as long as you're speaking? The record LED should stay lit, then when you release both LED's will be lit (processing) and when Alexa speaks only the playback LED will be lit.

twitherington, 2016-05-23:
the record LED stays lit as I keep the button pressed, but when I release it the lights flash back and forth, but I don't get any feedback from Alexa.

I'm gonna flash my sd card and start from scratch. I'll post here if that resolves the issue.

twitherington, 2016-06-18:
Issue resolved itself after reformatting, cloning repo, and adding the new IP and port (http://192.168.1.7:5000) to my Amazon Voice Services > Security Profile > Web Settings - Allowed Origins, and IP:port/code (http://192.168.1.7:5000/code) to Allowed Return URLs. After running ./setup.py and rebooting everything was working as it should.

closed 2016-06-18 by twitherington