alexa-pi/AlexaPiDEPRECATED #88: Bluetooth headset

Tiqni, 2016-05-21:
Anyone tried to use Bluetooth headset including speaker and microphone?

On my side only the speakers (work with A2DP), the microphone doesn't appear with arecord -l

It seems that HFP/HSP support is not fully working at any side (ALSA, PulseAudio, Bluez...)

chilinski, 2016-05-23:
I had the same results. Speaker worked, microphone did not. There's a new version of Jessie out and that makes me want to try again.

chilinski, 2016-05-24:
I want to add a note here. The lastest (May 2016) Jessie makes it much much easier to connect to bluetooth devices. Everything needed is installed by default. To set up bluetooth speakers involves clicking the Bluetooth manager button at the top of the screen and selecting and pairing the device and then clicking the audio button to choose the default speaker. But I still can't get the mic on the headset to work. I believe the issue is needing the proper file in /lib/firmware/brcm because dmesg complains the file is missing. The Plugable site says this firmware file is needed to get HFP/HSP. I've downloaded and put files in there, but they break the IOGear bluetooth adapter I have.

chilinski, 2016-06-01:
I got this response yesterday from Plugable:
"As you have discovered, Jessie in the Pi does not support headset profiles at the moment--at least we have not been able to get it to work here either. HSP/HFP has changed in recent kernel versions, and I think that is the reason. I have been planning to figure out how to do it (if possible) and write a blog post on it, but haven't been able to do it yet."
When I suggested the built in bluetooth on the Pi3 might be the answer, Plugable said, "Actually, I'm pretty sure HSP/HFP (the headset profile) doesn't work with the built-in chip either. If the Pi had support for the headset profile, it should be able to work with any adapter, including ours. I think the problem is that the Pi doesn't have support, and this I think is because support is complicated right now, and their team probably hasn't gotten around to including it."

Tiqni, 2016-08-25:
I spent again several days trying all possible packages and tricks, but I couldn't make the Bluetooth microphone appear in the arecord -l devices.

By the way, if someone can explain us how the **oFono** stack works, I found no documentation...

@chilinski Thanks for Plugable feedback, it makes it more clear that HFP/HSP are not at all supported for the moment. So I give up and wait for an official re-support of those profiles.

Tiqni, 2016-08-25:
I close the issue since it is now not directly linked to @AlexaPi.

Tiqni, 2016-08-25:
I close the issue since it is now not directly linked to AlexaPi.

closed 2016-08-25 by Tiqni