alexa-pi/AlexaPiDEPRECATED #89: Bad sound quality of Alexa

Tiqni, 2016-05-21:
I redirect audio output to my USB headset.

The "hello.mp3" message is OK, but Alexa answers aren't, the sound is crackling and like corrupted.
I don't see why since we are using the same parameters in play_audio() function.

(HDMI and jack 3.5mm are both OK)

korniza, 2016-05-31:
I get also the same issue. choppy sound with the following issues. I use Raspberry Pi 2 with jack 3.5 output. I suspect it has to do with internet speed / cache...

```
State.Playing
set player
[025a1528] alsa audio output error: cannot estimate delay: Input/output error
[025a1528] alsa audio output error: cannot estimate delay: Input/output error
[025a1528] alsa audio output error: cannot estimate delay: Input/output error
[025a1528] alsa audio output error: cannot estimate delay: Input/output error
```

bdatdo0601, 2016-07-03:
@korniza is there any fix on yours? It's really weird that l get this error almost everytime before but after install this:
`sudo apt-get install libsox-fmt-mp3`
The frequency of me getting it is less

bdatdo0601, 2016-07-07:
@korniza I finally solved the problem. It was  not because of the speed. Rather you just need to change vlc.instance('--aout=alsa') to vlc.instance().

luisbarreto, 2016-07-17:
@bdatdo0601 Can you please specify where the change you're proposing should be made?

swilso793, 2016-08-27:
In main.py

renekliment, 2016-10-26:
Seems solved -> closing. Reopen if needed.

closed 2016-10-26 by renekliment

ChristopherMori, 2017-03-22:
Noob here... can someone detail how to fix this issue in better detail? :(

renekliment, 2017-03-22:
@ChristopherMori This project is closed. Use https://github.com/alexa-pi/AlexaPi ... if you have issues there, try the _sox_ playback handler. More info in the docs here / chat on Gitter.