alexa-pi/AlexaPiDEPRECATED #92: Not able to capture input

architv, 2016-05-27:


chilinski, 2016-06-01:
On two recent projects, I created some really strange results using push button switches. On one with an Arduino, I accidentally swapped a NO switch with a NC switch. Since I was looking for a "change" and not just on or off, it really messed things up (the software didn't care if the switch was on or off, just that it had changed state from on to off or off to on). Also,  on a button box for sim racing I made, I overheated a switch while soldering the leads to it and that changed it from NO to NC.

renekliment, 2016-10-26:
Do you still have issues with this?

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment