alexa-pi/AlexaPiDEPRECATED #93: Logitech Webcam button usage

dlarue, 2016-05-30:
Have the original Master branch working with a USB Logitech webcam which has one button on it. I added a thread to the process to handle the blocking of the event. Here's the code snippet if anyone cares to also implement it as opposed to a GPIO button.

The following are most of the code snippets(if you know a little about coding and python you should be able to figure it out. But
ping me if you can't):

# USB webcam mic
import threading
from pygame import camera
from evdev import InputDevice, categorize, ecodes
from select import select

#
# USB webcam mic
# Setup
camera.init()
cam = camera.Camera("/dev/video0",(640,480))
cam.start()

# assign your own eventX identifier. I have no other USB devices connected so it's always 0
dev = InputDevice('/dev/input/event0') #logitech Camera

# threading class for button pushs
class ButtonValue(object):
    def __init__(self, start=0):
        self.lock = threading.Lock()
        self.value = start
    def down(self):
        self.lock.acquire()
        try:
            self.value = 1
        finally:
            self.lock.release()
    def up(self):
        self.lock.acquire()
        try:
            self.value = 0
        finally:
            self.lock.release()

def start(butVal):
    #last = GPIO.input(button)
    last = 0 if butVal.value else 1   # reverse from GPIO
  .....

def worker(c):
    """worker function"""
    print 'WebCam Button Worker Started'
    while True:
        r,w,x = select([dev], [], [])
        event = dev.read()
        for event in dev.read():
            if event.type == ecodes.EV_KEY:
                v=event.value
        if( v == 1 ):
            c.down()
        else:
            c.up()

if __name__ == "__main__":
#    try:
        butVal = ButtonValue()
        t = threading.Thread(target=worker, args=(butVal,))
        t.daemon=True # cause thread to terminate with main
        t.start()
.....
        start(butVal)
        t.join()
        sys.exit()

renekliment, 2016-10-26:
This sounds great. Would you care to implement is as a new _device_plaftorm_ over at the new upstream (https://github.com/alexa-pi/AlexaPi)?

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment