alexa-pi/AlexaPiDEPRECATED #95: unable to play audio file in response to question "what's new?"

marcoeg, 2016-06-10:
Error in playing back the audio file received in response to the "what's new?" question.
The log below indicates a problem with the playback. Perhaps because of file size or file name?

Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:Say:DAILYBRIEFING:DailyBriefingIntroduction", "audioContent": "cid:DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:Say:DAILYBRIEFING:DailyBriefingIntroduction_696261138"}}, {"namespace": "AudioPlayer", "name": "play", "payload": {"playBehavior": "REPLACE_PREVIOUS", "audioItem": {"streams": [{"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 0}, "streamUrl": "cid:DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelIntroduction:0", "offsetInMilliseconds": 0, "expiryTime": null, "streamId": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelIntroduction:0", "progressReportRequired": false}, {"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 0}, "streamUrl": "cid:DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelItem:0:0", "offsetInMilliseconds": 0, "expiryTime": null, "streamId": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelItem:0:0", "progressReportRequired": false}], "audioItemId": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelItem:0:0"}, "navigationToken": "amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#ssml#ACRI#DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelItem:0:0"}}]}, "messageHeader": {}}
Play_Audio Request for: /home/pi/AlexaPi/tmpcontent/DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:Say:DAILYBRIEFING:DailyBriefingIntroduction_696261138.mp3
Player State: State.Opening
unable to connect to localhost:3389
Error: protocol security negotiation or connection failure
[74506058] rdp demux error: can't connect to rdp server
[023cf078] core input error: open of `/home/pi/AlexaPi/tmpcontent/DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:Say:DAILYBRIEFING:DailyBriefingIntroduction_696261138.mp3' failed
[023cf078] core input error: Your input can't be opened
[023cf078] core input error: VLC is unable to open the MRL '/home/pi/AlexaPi/tmpcontent/DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:Say:DAILYBRIEFING:DailyBriefingIntroduction_696261138.mp3'. Check the log for details.
Player State: State.Error
Play_Audio Request for: /home/pi/AlexaPi/tmpcontent/DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelIntroduction:0.mp3
Play_Audio Request for: /home/pi/AlexaPi/tmpcontent/DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelItem:0:0.mp3
 Ready to Record.
Player State: State.Opening
unable to connect to localhost:3389
Error: protocol security negotiation or connection failure
[70100970] rdp demux error: can't connect to rdp server
[745c4fa0] core input error: open of `/home/pi/AlexaPi/tmpcontent/DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelIntroduction:0.mp3' failed
[745c4fa0] core input error: Your input can't be opened
[745c4fa0] core input error: VLC is unable to open the MRL '/home/pi/AlexaPi/tmpcontent/DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelIntroduction:0.mp3'. Check the log for details.
Player State: State.Error
Player State: State.Opening
unable to connect to localhost:3389
Error: protocol security negotiation or connection failure
[745b7698] rdp demux error: can't connect to rdp server
[71b0aca8] core input error: open of `/home/pi/AlexaPi/tmpcontent/DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelItem:0:0.mp3' failed
[71b0aca8] core input error: Your input can't be opened
[71b0aca8] core input error: VLC is unable to open the MRL '/home/pi/AlexaPi/tmpcontent/DailyBriefingPrompt.6be0bdf6-84f9-4c68-bf47-f30875e77a2f:ChannelItem:0:0.mp3'. Check the log for details.
Player State: State.Error

cyberdie, 2016-06-28:
SAme problem here.
Any help?

Play_Audio Request for: /root/AlexaPi/tmpcontent/DailyBriefingPrompt.49f87fb0-3a                                                                                                                               84-4926-bddc-5f754f10e356:Say:DAILYBRIEFING:DailyBriefingIntroduction_1998820366                                                                                                                               .mp3
Player State: State.Opening
unable to connect to localhost:3389
Error: protocol security negotiation or connection failure
[b5712800] rdp demux error: can't connect to rdp server
[019a74b8] core input error: open of `/root/AlexaPi/tmpcontent/DailyBriefingProm                                                                                                                               pt.49f87fb0-3a84-4926-bddc-5f754f10e356:Say:DAILYBRIEFING:DailyBriefingIntroduct                                                                                                                               ion_1998820366.mp3' failed
[019a74b8] core input error: Your input can't be opened
[019a74b8] core input error: VLC is unable to open the MRL '/root/AlexaPi/tmpcon                                                                                                                               tent/DailyBriefingPrompt.49f87fb0-3a84-4926-bddc-5f754f10e356:Say:DAILYBRIEFING:                                                                                                                               DailyBriefingIntroduction_1998820366.mp3'. Check the log for details.
Player State: State.Error
Play_Audio Request for: /root/AlexaPi/tmpcontent/DailyBriefingPrompt.49f87fb0-3a                                                                                                                               84-4926-bddc-5f754f10e356:ChannelIntroduction:0.mp3
Play_Audio Request for: /root/AlexaPi/tmpcontent/DailyBriefingPrompt.49f87fb0-3a                                                                                                                               84-4926-bddc-5f754f10e356:ChannelItem:0:0.mp3
Ready to Record.
Player State: State.Opening
unable to connect to localhost:3389
Error: protocol security negotiation or connection failure
[b2c19978] rdp demux error: can't connect to rdp server
[b57a8650] core input error: open of `/root/AlexaPi/tmpcontent/DailyBriefingProm                                                                                                                               pt.49f87fb0-3a84-4926-bddc-5f754f10e356:ChannelIntroduction:0.mp3' failed
[b57a8650] core input error: Your input can't be opened
[b57a8650] core input error: VLC is unable to open the MRL '/root/AlexaPi/tmpcon                                                                                                                               tent/DailyBriefingPrompt.49f87fb0-3a84-4926-bddc-5f754f10e356:ChannelIntroductio                                                                                                                               n:0.mp3'. Check the log for details.
Player State: State.Error
Player State: State.Opening
unable to connect to localhost:3389
Error: protocol security negotiation or connection failure
[b57abcb8] rdp demux error: can't connect to rdp server
[b2c22ad0] core input error: open of `/root/AlexaPi/tmpcontent/DailyBriefingProm                                                                                                                               pt.49f87fb0-3a84-4926-bddc-5f754f10e356:ChannelItem:0:0.mp3' failed
[b2c22ad0] core input error: Your input can't be opened
[b2c22ad0] core input error: VLC is unable to open the MRL '/root/AlexaPi/tmpcon                                                                                                                               tent/DailyBriefingPrompt.49f87fb0-3a84-4926-bddc-5f754f10e356:ChannelItem:0:0.mp                                                                                                                               3'. Check the log for details.
Player State: State.Error

renekliment, 2016-10-26:
You can give the new upstream (https://github.com/alexa-pi/AlexaPi) a try.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment