alexa-pi/AlexaPiDEPRECATED #97: Exit No response main.py

SharpBombsMC, 2016-06-11:
I got the flooie AlexaPi because I don't have a button, and I prefer not to.
I installed vlc: apt-get install vlc
I installed pyaudio: pip install pyaudio and apt 

I got my pmdl recorded from snowbound
I did: python main.py Alexa.pmdl

About 3 seconds it exits with no errors.the log says: "Starting Alexa..." then nothing.

Help please

Meskanen, 2016-06-11:
Exactly the same problem... I tried re-installing on  a fresh Raspberry, but still nothing. I also asked Flooie and he told me to ask here for more help.

SharpBombsMC, 2016-06-12:
If it helps I Installed VLC and pyaudio

maso27, 2016-06-13:
I've got nothing to offer as far as the flooie branch is concerned. But I have a working voice-commandable version of my own working at [https://github.com/maso27/AlexaPi](https://github.com/maso27/AlexaPi)

It uses Sphinx, so it's not as tightly configured as Snowboy. But it's been working well in my house for a while now.

It takes a while to build sphinx, but everything installs great on its own from setup.sh.

Caveat:
I added an option to install shairport-sync for airplay support, and that runs on port 5000, so I changed the authorization process to port 5050.  You'd have to change that in your developer settings unless you re-use an existing creds.py file.

Meskanen, 2016-06-13:
@maso27 Thank you! I installed your version and it works perfectly :)

SharpBombsMC, 2016-06-13:
@maso27 I am buying a new microphone soon. I +1 assuming it worked. +1 for responding anyways! :) Thanks!

SharpBombsMC, 2016-06-19:
@maso27 I tried yours. It doesn't work D: It says Player State: State.Ended and then nothing else... Is that supposed to happen?

maso27, 2016-06-19:
That's supposed to happen.  When AlexaPi first starts, it plays "hello.mp3" (which you should be able to hear if your audio is set up right), then that state ends and it waits for input.

So you should see:
```
Checking Internet Connection...
Connection OK
Play_Audio Request for: /root/AlexaPi/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
```

You should be able to then hold the button and say your command, or (in the voice-enabled branches) say "Alexa" and then say your command.

Do you hear her say "hello" when she starts up?

SharpBombsMC, 2016-06-19:
@maso27 when it wanted to say hello I heard a huge static sound.

I started it by doing this.

root@raspberrypi ~/AlexaPi $ python main.py

It does everything you said it should do but hello.mp3 is static?

I am in the directory root/AlexaPi

SharpBombsMC, 2016-06-19:
@masp27 what voice activated branch? I just did git clone and your repo.

maso27, 2016-06-19:
I was referring to mine and flooie's implementations when I referenced the voice-enabled branches.

So it sounds like you need to get your audio settings figured out. What are you using for your input and output? 3.5mm jack? HDMI? USB dongle?

Google yielded this page: http://blog.scphillips.com/posts/2013/01/sound-configuration-on-raspberry-pi-with-alsa/
Might be helpful.

flooie, 2016-06-19:
@SharpBombsMC - the huge static sound - is something I originally encountered myself.  I believe it is a setting issue on your raspberry pi.

SharpBombsMC, 2016-06-19:
USB Dongle. It is a StarTech.has a card. I will look at the website.

SharpBombsMC, 2016-06-19:
Also. Saying Alexa doesn't work
..

maso27, 2016-06-21:
@SharpBombsMC have you had any luck getting this to work?
I'd bet it just comes down to getting your audio input and output set up right.

I seem to remember that "sudo raspi-config" will allow you to force a default audio output device.
And you might play around with adjusting the "device" variable inside main.py.

SharpBombsMC, 2016-06-22:
@maso27 Nope. I still hear the static with hello.mp3 I will try that raspi-config

If I can't hear hello.mp3 would my mic still work? Aplay works...

maso27, 2016-06-22:
Mic and speaker are (relatively) unrelated.  Use "arecord -l" to list the input devices.  Then you should be able to use the name of your mic from that in the "device" variable within main.py.
I say "should" because I haven't ever had to change mine.  

I've been using the 3.5mm jack for speakers, and a USB webcam or microphone for the input.  Since there's only one option, I don't have to change from defaults at all.
And honestly, people say the audio isn't good on the jack but I haven't yet found that to be my audio quality bottleneck.

SharpBombsMC, 2016-06-23:
@maso27 I will test it out today. If it helps my USB microphone has static whenever im not talking to it

SharpBombsMC, 2016-06-23:
@maso27 I really appreciate you helping me! If this helps you, my sound card is a a speaker and a mic. Well, I have to plug my AUX into the sound card. The microphone works when I do plughw 1,0 when I'm recording using a record. I haven't tried what you said yet but I hope it works!

SharpBombsMC, 2016-06-23:
@maso27 I found something!

`  File "main.py", line 379, in start
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: Device or resource busy
`

maso27, 2016-06-23:
Try "ps ax | grep main.py" and see if Alexa's running more than once. (So you should see just one "python /root/AlexaPi/main.py" and then a line with "grep" on it)

Maybe Alexa is double-loading.  If there's more than one instance, type "kill xxxx" where xxxx is the number in front of one of the lines above.

SharpBombsMC, 2016-06-23:
ok! Also, alexa might not be recognizing my voice because my microphone is being static. It is fixed by doin:  -f S16_LE -r 44100 in arecord. Can I add that to main.py somehow?

SharpBombsMC, 2016-06-23:
@maso27 I get `2596 pts/0    S+     0:00 grep main.py
2596` I restarted the pi and still the same error. :/

maso27, 2016-06-23:
I'm not sure I understand.
Type _**ps ax | grep main.py**_

If things are good, you should see:
_**384 ?        Sl     0:15 python /root/AlexaPi/main.py
13917 pts/0    S+     0:00 grep --color=auto main.py**_
...or very similar.

If you see the first line twice, (with different numbers), type _**kill 384**_ or whatever the number is for one of the running python instances.
That will kill one instance of main.py, keeping the other one running.

Then try playing with Alexa (no reboot).  If that works, you'll have to figure out why main.py is starting more than once on reboot and fix it.

As to the static--the script sets the microphone for the format Alexa wants it to use.  So you changing the format would break Alexa's ability to listen to the files.  I'd see if killing an extra process helps before moving on to that.

SharpBombsMC, 2016-06-23:
@maso27 now how am I supposed to run main.py and grep?

Main.py: https://gyazo.com/5c5011972a610df59069e0c9159f5b4d

Grep: https://gyazo.com/dc2b82b9050d630d10414d40b5a784f1

maso27, 2016-06-23:
Aaah.  You're running it from the terminal.

So the easiest way, which is how this runs by default, is to type:
_**python main.py &**_

The ampersand tells it to run in the background.  Then you can use the terminal for other things, and it'll keep running after you log out.

Try that.
But based on what you just showed me, it looks like it doesn't recognize your microphone.

If you don't have more than one instance of main.py running, type _**arecord -l**_ and see what is returned in the list of capture devices.  If your microphone is anything other than card 1, you may need to change the "device" variable within main.py.

SharpBombsMC, 2016-06-23:
@maso27 I re cloned the repo and set it up. Now this happens:

I do python main.py and it goes: 
`Checking Internet Connection...
Connection OK
`

thats it. It says hello.

illperipherals, 2016-06-23:
Are you still on flooie's voice branch?

SharpBombsMC, 2016-06-23:
No, i am not still in his branch.

SharpBombsMC, 2016-06-23:
Now this happens: https://gyazo.com/6e9c8efa4bea57c76fffe3f0113bce22

... It is not double starting.

maso27, 2016-06-23:
@SharpBombsMC 
I added command line switches yesterday that enable "silent" mode and "debug" mode.
Before that, "debug" mode was on by default, which spewed out all the other lines of information.  Now you can enable it by typing _**python main.py -d &**_ to start it.

Silent mode will turn off the "hello" when it boots.

@illperipherals 
We're working on the branch I made, which is a simpler voice-enabled implementation.

SharpBombsMC, 2016-06-23:
It now works! thank you!

SharpBombsMC, 2016-06-23:
Never mind D: Now it is giving the error again :(

illperipherals, 2016-06-23:
Thanks @maso27 . I had PocketSphinx working, but just started to move over to snowboy.
I am trying to get Telegram working with the PicoTTS for testing. I have it loading up in yakuake right now that will feed voice files over. Is there an IRC chan?

@SharpBombsMC , are you using the single USB for in/out? (didn't mean to interrupt your issue)

flooie, 2016-06-23:
For what its worth - I modified and built my code on a RPi 3.

maso27, 2016-06-23:
@flooie 
I love the work you've done!  I've found configuration of snowboy to be a bit laborious, so I haven't embraced it yet.  I'm holding off until things get a bit more... automatic?

I've been alternating between a RPi 2 and RPi 3, depending on whether I'm at home or work.

SharpBombsMC, 2016-06-23:
@maso27 @flooie Sometimes (most of the time) I will get static as an answer... Why? How can I fix this?

maso27, 2016-06-23:
@SharpBombsMC I'm not sure on that one.  Maybe try the onboard 3.5mm jack?  You'll have to change the default output for that though.

SharpBombsMC, 2016-06-23:
I don't know how to change it for the 3.5mm jack. How do I do that? It always wants to come out my sound card. My sound card has a jack built in. I want it to come out my default raspberry pi jack. How do I do that?

illperipherals, 2016-06-23:
@SharpBombsMC can you go through the details when playing directly to the USB?
something like: aplay myfile.wav -v -V mono -D sysdefault:CARD=1

flooie, 2016-06-23:
Have you been able to play anything. Unrelated to this project.  aplay style?

illperipherals, 2016-06-23:
auto is "0", 3.5 is "1", HDMI is "2"l

SharpBombsMC, 2016-06-23:
@flooie Yes! Sometimes it will be static though. It never was static until had to plug it in to my sound card. I have a raspberry pi 3. I want my sound card JUST FOR MY MIC. and my default audio jack 3.5mm just for speakers. How would I do that?

SharpBombsMC, 2016-06-23:
Inside .asoundrc I have it set to record my sound card as default. PCM or something. When I plug my AUX into my default not sound card audio jack nothing is heard. When I plug it into my sound card audio jack I hear static.

SharpBombsMC, 2016-06-23:
@maso27 since this project is set to record from plughw one (USB) I don't have to set my mic as default right? Then everything works if so

maso27, 2016-06-23:
If your mic is the only capture device you have attached, you shouldn't need to set it as default.

SharpBombsMC, 2016-06-23:
I will try that and get back to you later!

Thank you all for replying and helping I really appreciate it.

@maso27 @flooie @illperipherals

SharpBombsMC, 2016-06-23:
Everything works! Thank you so much! It has always been my dream for this!

shahvineet98, 2016-08-05:
@maso27 
[Traceback (most recent call last):
  File "main.py", line 84, in <module>
    decoder = Decoder(config)
  File "/usr/local/lib/python2.7/dist-packages/pocketsphinx/pocketsphinx.py", line 271, in __init__
    this = _pocketsphinx.new_Decoder(*args)
RuntimeError: new_Decoder returned -1
](url)

I am getting the following error when I run `python main.py`. Any idea how to resolve this?

maso27, 2016-08-05:
@shahvineet98 
That's kind of a whole new issue.  
But I can reproduce the error by removing the "pocketsphinx" directory from /root.

Try typing the following:
```
cd ~
git clone https://github.com/cmusphinx/pocketsphinx.git
```

Danthelord98, 2016-08-13:
@maso27 
I have exactly the same issue as @shahvineet98.

The fix you suggested unfortunately does not fix the problem. Whenever I run main.py I get exactly the same error code.

renekliment, 2016-10-26:
Seems solved -> closing. Reopen if needed.

closed 2016-10-26 by renekliment