alexa-pi/AlexaPiDEPRECATED #98: Updating to amazon voice api v2

amahlaka, 2016-06-19:
Hi, i noticed that timers dont work, and some other stuff as well, looks like this verison is using old v1 api, will this be migrating to apiv2 anytime soon?
here is migration guide https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/docs/migration-guide
@sammachin

sammachin, 2016-06-21:
Yep its on my list of things to do but I've also been working to cleanup the codebase and integrate the music playback so I don't know how long it will take me to find the time to migrate. The migration isn't simple as its HTTP/2 based.

parttimehacker, 2016-07-11:
I know you are busy but this is high on my list. I've got the current version working (thank you!), but Amazon has the rev 2 out which requires html/2 as you know.

sammachin, 2016-09-01:
I've finally had some time to look at this project, I've spent the last week refactoring the code to be a lot more modular so that we can support multiple different setups and different ways of triggering eg, button, Spinx, Snowboy etc. This should allow me/somebody to work on the v2 api as well without breaking the rest of the project.

closed 2016-09-01 by sammachin