alexa-pi/AlexaPiDEPRECATED #100: Amazon instructions vs. AlexaPi Implementation

caiismyname, 2016-06-23:
Hi all! (Heads up -- not an issue, just a question)

I was trying to create my own version of an Alexa interface on RPi through Python (for learning!) using this implementation as an example. I noticed that on Amazon's [instructions](https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/docs/managing-an-http-2-connection) regarding interfacing with Alexa, they say to open and maintain an HTTP/2 connection that remains open through the lifetime of the connection and is used to send/deliver events/directives. 

However, in this AlexaPi implementation, it seems as though that step is omitted, and every time an command is given, the audio file is directly sent to the speech recognizer `'https://access-alexa-na.amazon.com/v1/avs/speechrecognizer/recognize'`. 

Could someone explain why the constantly open connection is skipped? Thanks!

lennysh, 2016-06-23:
You're referring to the second version of the Amazon API with the open connection. This script currently is using the first version of the Amazon API.

caiismyname, 2016-06-23:
Oh, that explains it! Thanks. Do you know of any working examples using the 2nd API version? I'm having trouble establishing that connection

flooie, 2016-06-23:
There is a work in progress project on github for v2.

flooie, 2016-06-23:
Also I believe a few of us are working on it.  I'm working on an objc version as well.  Also the v2 is for python 3

caiismyname, 2016-06-24:
Is that work in progress public? Would you mind pointing me towards it so I can have a general guide to look at? Thanks!

gregwa1953, 2016-06-24:
@David Cai,
"Works in progress" are rarely public.  That's because some people would
try to build it without it being finished and be very frustrated.

sammachin, 2016-09-01:
I've finally had some time to look at this project, I've spent the last week refactoring the code to be a lot more modular so that we can support multiple different setups and different ways of triggering eg, button, Spinx, Snowboy etc. This should allow me/somebody to work on the v2 api as well without breaking the rest of the project.

closed 2016-09-01 by sammachin

renekliment, 2016-09-09:
> There is a work in progress project on github for v2.

@flooie Would you care to elaborate? Is this still relevant? We're currently doing a rewrite of AlexaPi and it would be a shame to do something that's been done already. Thanks

flooie, 2016-09-10:
@renekliment 

I built an objective-c version as an extension of the work here that takes both voice commands and text inputs (so I can give it commands coffee shops).  I wasn't sure if was appropriate to let y'all know about it here in the python project.  It works pretty well just need to add your own client_id - token - and secret.  Would love help improving it.  

https://github.com/flooie/Mecho