alexa-pi/AlexaPiDEPRECATED #99: Outline of differences and similarities with official alexa-avs-raspberry-pi project

Thelle, 2016-06-23:
Really cool project - but it is a little hard for me to see the main differences between this project and the official Amazon project https://github.com/amzn/alexa-avs-raspberry-pi
Maybe to an outline of the main points in the ReadMe-file?

dlarue, 2016-06-23:
Here's a few I've noticed:
It'll run on a full OS( Linux and maybe even Windows ).
It's written in Python which is what makes it so portable.
It's already designed to run on one of the most popular embedded boards( rPi ).
You won't find it in an app store.

rahulpathakgit, 2016-07-05:
It supports HTTP/2 API version

sammachin, 2016-09-01:
For me the main difference with the 'official' project is that one uses a keyboard/mouse and screen for the user interface eg you have to click a button on a gui to activate Alexa. This project uses hardware buttons and LED's. Also the auth on AlexaPi is  web based.

This means that you can use the RasPi more like an applicance just like the Echo or Tap not like a computer.

As mentioned above it does support the v2 API and thats on the list of things I want to get working in AlexaPi, Amazon have slightly more time and resources than I do to work on stuff!

Oh and theirs is written in Java, mine is Python, different people like different things :)

closed 2016-09-01 by sammachin