alexa-pi/AlexaPiDEPRECATED #104: Error when switching on Alexa

ankush1012, 2016-07-14:
Hi,
I am getting the below error as soon as I'm switching on button to talk to Alexa and program is ending.

Traceback (most recent call last):
File "main.py", line 156, in <module>
start()
File "main.py", line 135, in start
l, data = inp.read()
UnboundLocalError: local variable 'inp' referenced before assignment

renekliment, 2016-10-26:
This is weird. Are you using latest code from the repo?

illperipherals, 2016-10-26:
This has come up before: https://github.com/sammachin/AlexaPi/issues/13

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment