alexa-pi/AlexaPiDEPRECATED #105: response 204

stu0219, 2016-07-18:
Hi :
I've try the demonstration, but there is no response after I recorded
So I print out the r.status_code is 204, which mean no content, Is any way to figure out reason??

Thanks

sammachin, 2016-07-18:
Usually this is because the audio file that was sent to alexa is blank/silent.

The last recorded command should be in recording.wav, try playing that and see what it sounds like, could be an issue with your mic

stu0219, 2016-07-18:
Thanks, I found the answer, and the problem is cause permission error
, and it worked right now

davebel, 2016-08-05:
Hi, I'm having this issue too. I can play recording.wav and it sounds clear. Can you let me know how you resolved it? Thanks

yangsuper, 2016-10-13:
@stu0219 Hi, can you please share the solution you tried ? I hit this issue as well...

renekliment, 2016-10-26:
@yangsuper You can give the new upstream a try (https://github.com/alexa-pi/AlexaPi). If there's some permissions error here, it should be resolved over there.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment