alexa-pi/AlexaPiDEPRECATED #106: no response after "Connection OK"

JinheeC, 2016-07-18:
Hello, I got refresh_token and now I tried to start main.py.
when I type "python main.py" it says "checking internet connection" and "Connection OK".
but there is no response after these statements.

I tried almost things of the previous answers related to my question.
1. I changed the device variable from "plughw:1" to "Device"
2. I changed the setperiodsize(500)

but, nothing was changed..

stu0219, 2016-07-18:
Print out the r.status_code at alexa() below, at least know pi is triggered or not

luisbarreto, 2016-07-18:
Hello, same issue here.
Printing the r.status_code gives a 403 witch is "Forbiden" / "Unauthorized"

I double checked the creds.py and every field is correctly filled (double checked with copy/paste) with my credentials.
The only one empty is the refresh_token = '' witch is blank.

Any pointers would be greatly appreciated.

stu0219, 2016-07-19:
It might be not same issue
Attribute refresh_token should not be empty, there is mistakes of your setup progress
When the setup be done, refresh_token of cred.py need to full of string, try adding sudo maybe?

JinheeC, 2016-07-19:
@stu0219 thanks, I printed the status_code and it was 200 or sometimes 204.
I think it works properly. but I can't hear anything from my earphone.

closed 2016-07-19 by JinheeC