alexa-pi/AlexaPiDEPRECATED #108: I've been running version1.1 and it seems to work fine until it breaks

fernando-vera, 2016-07-26:
Hi,

First I want to thank you for porting this over I know its a lot of work. I wanted to give some feedback. It looks like radio is working with this version. As you said in another comment timers and alarms aren't supported on this version of the API. I did get something odd happen. Alexa will work fine for maybe 15 minutes playing around with it then it will say "I'm sorry I'm having trouble please try in a little while"

This is what gets printed to the terminal:
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.#ACRI#ErrorNotifier_cd6f174c-8722-4fa3-a104-92c4f56a68db", "audioContent": "cid:ErrorNotifier_cd6f174c-8722-4fa3-a104-92c4f56a68db_1589464909"}}]}, "messageHeader": {}}
Play_Audio Request for: /home/pi/Downloads/AlexaPi/tmpcontent/ErrorNotifier_cd6f174c-8722-4fa3-a104-92c4f56a68db_1589464909.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Ready to Record.

I tried this both on a raspberry pi 2 and 3. On more than one network. It has happened consistently every time I run it. It will give the same error every-time after I begin getting it and I have managed to clear it by reinstalling Alex although this doesn't always fix it. I think it may have to do something with running an app. I tried running two apps a game show app and the daily show app. It will work fine some time after running the app so its not clear if that is the cause. I have tried hard and soft resets but it doesn't seem to get rid of the problem. I'm still new to the raspberry pi ecosystem and new to python so I'll try to see if I figure something out although I'm hoping someone can help me out here.

Thank you!

fernando-vera, 2016-07-26:
Update: It's very odd it started working again after a restart although that's not always the case. It worked fine then I ran the teacher skill and it worked but after finishing it no longer worked and gave me the same error. I think there may be a bug with Alexa skill integration? I'm hopping someone else can tell me if they have the same issue with version1.1.

renekliment, 2016-10-26:
Can you see if it works in the new upstream (https://github.com/alexa-pi/AlexaPi)? If we're fixing it, it would most probably be over there. Thanks!

renekliment, 2017-01-05:
This has been fixed over at the new repo, closing.

closed 2017-01-05 by renekliment