alexa-pi/AlexaPiDEPRECATED #109: Amazon authentication fails

hemingweb, 2016-07-27:
I'm unable to authenticate with amazon. I get an error related to the Allowed Return URL, I've added 'https://localhost:5000/authresponse' and 'https://192.168.25.7:5000/authresponse'.

When I call the auth_web.py I get the following output:

sudo python auth_web.py
[27/Jul/2016:18:33:14] ENGINE Listening for SIGHUP.
[27/Jul/2016:18:33:14] ENGINE Listening for SIGTERM.
[27/Jul/2016:18:33:14] ENGINE Listening for SIGUSR1.
[27/Jul/2016:18:33:14] ENGINE Bus STARTING
CherryPy Checker:
The Application mounted at '' has an empty config.

[27/Jul/2016:18:33:14] ENGINE Started monitor thread 'Autoreloader'.
[27/Jul/2016:18:33:14] ENGINE Started monitor thread '_TimeoutMonitor'.
[27/Jul/2016:18:33:14] ENGINE Serving on http://0.0.0.0:5000
[27/Jul/2016:18:33:14] ENGINE Bus STARTED 

I'm not sure where to start looking to sort this out, not sure if it's even and issue "The Application mounted at '' has an empty config." this seems odd as well "ENGINE Serving on http://0.0.0.0:5000"

Any help would be great.

Mavrikant, 2016-07-27:
Are you opening URL with your computer's browser or Raspberry pi's browser?

hemingweb, 2016-07-27:
I'm openinghte browser on the raspberry pi. I'm using a pi3, forgot to mention in the original post

hemingweb, 2016-07-28:
Here's a screenshot of the error
[image]

[image]

maso27, 2016-07-28:
As far as I know, "authresponse" is used in Amazon's demo app, not AlexaPi.
You want to replace "authresponse" with "code"

So 'https://localhost:5000/code' and 'https://192.168.25.7:5000/code'

(Also, in my setup I have http instead of https, but I don't think that makes a difference.)

hemingweb, 2016-07-28:
Epic. I changed 'autoresponse' to 'code' not sure where I missed that. the authentication failed again. I then changed https to http and tried again. Success, the service authenticated and prompted a reboot.

Cheers

closed 2016-07-28 by hemingweb