alexa-pi/AlexaPiDEPRECATED #111: Client is not authorised to connect to Serverxbc_connection_has_error()

hemingweb, 2016-07-29:
Hi,

Where would I starting to diagnose this issue?

Cheers

[image]

hemingweb, 2016-07-29:
I had installed zero conf and connect to the pi using screen sharing on my mac. Not sure if it relevant, is this a tightvnc issue?

maso27, 2016-07-29:
I've never seen that error, but it might be worth trying to use SSH instead of tightvnc.

AlexaPi runs fine on Debian Jessie Lite, so I've never tried it with a desktop environment.
You could see if it happens when you ssh in as root and then run it.

hemingweb, 2016-07-30:
Thank you for your help 

I don't get the this error using shh

closed 2016-07-30 by hemingweb