alexa-pi/AlexaPiDEPRECATED #113: 400 bad request

moorlag, 2016-08-02:
Hello, 

I'm having a problem; 400 bad request. 

We're sorry!
An error occurred when we tried to process your request. Rest assured, we're already working on the problem and expect to resolve it shortly.

> 400 Bad Request
> The redirect URI you provided has not been whitelisted for your application. Please add your redirect URI in the 'Allowed Return URLs' section under 'Web Settings' for your Security Profile on Amazon Developer Portal.
> Request Details
> scope=alexa%3Aall
> redirect_uri=http%3A%2F%2F192.168.10.95%3A5000%2Fcode
> response_type=code
> client_id=amzn1.application-oa2-client.c17b846ecfbc40a08b67bd7dcc102391
> scope_data=%7B%22alexa%3Aall%22%3A+%7B%22productInstanceAttributes%22%3A+%7B%22deviceSerialNumber%22%3A+%22001%22%7D%2C+%22productID%22%3A+%22my_device%22%7D%7D

moorlag, 2016-08-02:
Direct after giving the details from the portal this comes to my terminal. 

> Object "=" is unknown, try "ip help".
> [02/Aug/2016:18:26:06] ENGINE Listening for SIGHUP.
> [02/Aug/2016:18:26:06] ENGINE Listening for SIGTERM.
> [02/Aug/2016:18:26:06] ENGINE Listening for SIGUSR1.
> [02/Aug/2016:18:26:06] ENGINE Bus STARTING
> CherryPy Checker:
> The Application mounted at '' has an empty config.
> 
> [02/Aug/2016:18:26:06] ENGINE Started monitor thread '_TimeoutMonitor'.
> [02/Aug/2016:18:26:06] ENGINE Started monitor thread 'Autoreloader'.
> [02/Aug/2016:18:26:06] ENGINE Serving on http://0.0.0.0:5000
> [02/Aug/2016:18:26:06] ENGINE Bus STARTED
>

sammachin, 2016-09-01:
Does the 'Allowed Return URL' address have `/code` at the end, it should be a slightly different URL to the 'Allowed Origin'

lion6230i, 2016-10-18:
Hello,

I have the same error.

> Error Summary
400 Bad Request
The redirect URI you provided has not been whitelisted for your application. Please add your redirect URI in the 'Allowed Return URLs' section under 'Web Settings' for your Security Profile on Amazon Developer Portal.
Request Details
scope=alexa%3Aall
redirect_uri=http%3A%2F%2F0.0.0.0%3A3000%2Fcode
response_type=code
client_id=amzn1.application-oa2-client.1f12f17067fa44b9a149347afeb82449
scope_data=%7B%22alexa%3Aall%22%3A+%7B%22productInstanceAttributes%22%3A+%7B%22deviceSerialNumber%22%3A+%22001%22%7D%2C+%22productID%22%3A+%22my_device%22%7D%7D

I have change 
Allowed Origins   to  http://localhost:3000/code
Allowed Return URLs   to http://localhost:3000/code
But still can't work.

Any idea for this problem?

maso27, 2016-10-18:
AlexaPi doesn't use port 3000.  It uses either 5000 or 5050, depending on which version you're using.

Also, the only place "localhost" will work is if you're using the browser from within the Raspberry Pi.  If you're doing that it'll be fine.  But if not, better to put the actual IP address of the pi.

lion6230i, 2016-10-19:
@maso27 I change setting to 5000 (version 1.1) and use localhost as IP to browse it. 
It's display another error.

> 500 Internal Server Error
The server encountered an unexpected condition which prevented it from fulfilling the request.
Traceback (most recent call last):
  File "/usr/lib/python2.6/site-packages/cherrypy/_cprequest.py", line 670, in respond
    response.body = self.handler()
  File "/usr/lib/python2.6/site-packages/cherrypy/lib/encoding.py", line 220, in __call__
    self.body = self.oldhandler(*args, **kwargs)
  File "/usr/lib/python2.6/site-packages/cherrypy/_cpdispatch.py", line 60, in __call__
    return self.callable(*self.args, **self.kwargs)
  File "auth_web.py", line 35, in code
    line = 'refresh_token = "{}"'.format(resp['refresh_token'])
ValueError: zero length field name in format

Actually I use my PC(x86) to test this . Are those error cause by different platform?

renekliment, 2016-10-26:
If you want to run it on your PC, try the new upstream: https://github.com/alexa-pi/AlexaPi This repo doesn't have direct support for it. Get back to us then so we can see if we can close this :) Thanks!

lion6230i, 2016-10-27:
Hi renekliment,

I try to debug with this error message, so I print resp value

> 127.0.0.1 - - [26/Oct/2016:10:58:46] "GET / HTTP/1.1" 303 758 "" "Mozilla/5.0 (X11; Linux i686) AppleWebKit/537.36 (KHTML, li
ke Gecko) Chrome/45.0.2454.85 Safari/537.36"
127.0.0.1 - - [26/Oct/2016:10:58:47] "GET / HTTP/1.1" 303 758 "" "Mozilla/5.0 (X11; Linux i686) AppleWebKit/537.36 (KHTML, li
ke Gecko) Chrome/45.0.2454.85 Safari/537.36"
{u'access_token': u'Atza|IwEBIGmCbiWK9BPUDLy4qDOiz9hl0T_cldTl-C2GBruq7ExnCkEkZ2XQbcK5NIeJu63XTu3aYyOcQGZhNdEVfOv-kszZzgjVVX9fOSu2tUg49qz7nkueBQODgDqQHUwwovJGrIcvI9gwU3NjKUFK4_bDDhA6H1G409PYOXOkagQLjiylgPsrnNSWDzTG0VvV8KMYVDIvUbPfBTS-Zv3DacZMlK_7wxRPQR47s_Gu2kYkgTg_2wdYY3HT9G_rGvzh_zU1xeh8uWrWnyGBnn3AiYRyDb_G5s7xguWDqYDLGL7gcQi0qJrgZLUSo_OHn7Rsnv2xHfZGu6VlV8Q53XW0Wv-gTmBElLKmxphjvOG4Nhclxjy8h5Qj7cpeGPuuYCQTx_MiGXhJMbNdg44EJLob5I9PT3MS0wwkGN3ZLqlFB5a-oJHPZfb8lxuKz8lItXOYdKwQTFfLe3ngKIOV8E3_eyghT2XOzYVNmC-57F9BwBgzcugcBCBczkqAF25G1lbexLUaSNWHnE7wztF2QK-0WJlWRu4V', u'token_type': u'bearer', u'expires_in': 3600, u'refresh_token': u'Atzr|IwEBIGnguWIe1dgfqAUAD_EPFvrrqDTpyKF9chfzUnfrMG7YRZZuxARganYzhpDWVs0ux2VrAv05z4j3kxvOVqfuXwXFOX6N9GYE_N5huHjQRlfkjp9owfy0ZnvbqIP83W6emvzSNj-m7FthVSmZZuU4OP3VUM1snrp8szZpHIoqu85OVbHMDoQUHWC0ls4Lo__8vef8spuIen8yX3F6lUFd7I3Mgi5BwdOaDY24G9qcu405Ron9jrc29N_hhN16z9icjufk2YKB3grCsqiHM6paBrEwGjQskCxljR9VUIhq0E_0ArlFdtm72m-9WFq014_lRZgmNzFJZ-I8n-4xSZU6bYqoV2Powe_4V3e1YEaHSeJdQPzTy0gOWfH7abLEVw8SfnCs_0kDLUMH04hvkWFR_iJObhDve7qRwI_BRpLNmiBWTvk4KVNQFyEt7nmUT4zPVleOrsEi6MpXAwAem8kMZBuYjnun9lICxS06q-nRuzrlqb2y0JxIoH5SARgIHbAA7J-74Eb02cYhHerI5i35iGlu'}
[26/Oct/2016:10:58:56] HTTP
Traceback (most recent call last):
  File "/usr/lib/python2.6/site-packages/cherrypy/_cprequest.py", line 670, in respond
    response.body = self.handler()
  File "/usr/lib/python2.6/site-packages/cherrypy/lib/encoding.py", line 220, in __call__
    self.body = self.oldhandler(*args, **kwargs)
  File "/usr/lib/python2.6/site-packages/cherrypy/_cpdispatch.py", line 60, in __call__
    return self.callable(*self.args, **self.kwargs)
  File "auth_web.py", line 36, in code
    line = 'refresh_token = "{}"'.format(resp['refresh_token'])
ValueError: zero length field name in format
[26/Oct/2016:10:58:56] HTTP
Request Headers:
  CONNECTION: keep-alive
  COOKIE: gsScrollPos=0
  HOST: localhost:5000
  UPGRADE-INSECURE-REQUESTS: 1
  Remote-Addr: 127.0.0.1
  ACCEPT: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
  USER-AGENT: Mozilla/5.0 (X11; Linux i686) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/45.0.2454.85 Safari/537.36
  ACCEPT-LANGUAGE: zh-TW,zh;q=0.8,en-US;q=0.6,en;q=0.4,zh-CN;q=0.2
  ACCEPT-ENCODING: gzip, deflate, sdch

It's seems server have reply refresh token , but cherrypi module can't use format 
method to get it. Is it correct?

If it's correct, https://github.com/alexa-pi/AlexaPi  project also use cherrypi module
to get refresh token. Is it work?

renekliment, 2016-10-27:
Seems like the `refresh_token` is empty.
If you've got some time, just to be totally sure / confirm, please try the new upstream, according to the Installation Guide exactly - maybe even create new device at Amazon. There have been improvements in various places and one of them might have fixed it. Thanks!

lion6230i, 2016-10-27:
Hi renekliment,

Use https://github.com/alexa-pi/AlexaPi project have another problem.
> [root@localhost AlexaPi]# python src/auth_web.py 
/usr/lib/python2.6/site-packages/cryptography/__init__.py:26: DeprecationWarning: Python 2.6 is no longer supported by the Python core team, please upgrade your Python. A future version of cryptography will drop support for Python 2.6
  DeprecationWarning
Traceback (most recent call last):
  File "src/auth_web.py", line 54, in <module>
    print "Ready goto http://{}:5050 or http://localhost:5050  to begin the auth process".format(ip) 
ValueError: zero length field name in format

Could it cause by my python version too old?

renekliment, 2016-10-27:
It's very probably not caused by the python version. This is very peculiar - what OS (+ version) and device are you using?

lion6230i, 2016-10-27:
@renekliment 
CentOS release 6.7 (Final) 
PC(Intel(R) Core(TM)2 Quad CPU    Q8200  @ 2.33GHz)

illperipherals, 2016-10-27:
Yeah, just bring that up to date. I believe this is the issue:

https://docs.python.org/dev/library/string.html#format-string-syntax

renekliment, 2016-10-27:
So I've just compiled python 2.6 to confirm and it indeed seems to be the issue.
Alternatively, you can change that line to 
`print "Ready goto http://{0}:5050 or http://localhost:5050  to begin the auth process".format(ip)`, which works for me. 
However, we can't guarantee that other things will work as well. I guess we could change this thing, so it works, but in the long run I'm not sure we can (nor want to) support python 2.6 as it's super old.
Please upgrade to python 2.7.

lion6230i, 2016-10-27:
@renekliment 
It's work, but after login  LWA (Login With Amazon) . Web browser will show the information in below

> 500 Internal Server Error
The server encountered an unexpected condition which prevented it from fulfilling the request.

@illperipherals 
Right, I have try to get refresh_token without format method and it's work.

renekliment, 2016-10-27:
Yeah, that's what I was walking about. Unfortunately, I don't think we want to spend any more time on this, sorry. Please upgrade to python 2.7.

P.S.: If I were you, I'd upgrade to CentOS 7.

lion6230i, 2016-10-27:
@renekliment Which python version does you used to develop? In CentOS 6.7 I can find version 3.4 .

renekliment, 2016-10-27:
Python 3.x isn't yet supported. We'll hopefully add support soon. Stick to 2.7.x. for the time being.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment