alexa-pi/AlexaPiDEPRECATED #115: Connection Error

RobertSE, 2016-08-06:
Hi,

Having setup the project on the same 2015-11-21 Jessie Img on a RPi 2. I get repeat error messages on the alexa.log
"Checking Internet Connection
Connection Failed"

However I am connected to the internet (and can browse), what might be causing this issue?

I ran ./setup.sh provided the profile details, pointed my browser to http://localhost:5000, signed in, received the refresh token.
At which point I ^C the ./setup.sh command and rebooted the pi.

Have I done something wrong during setup?

Thanks.

shawpbx, 2016-08-25:
Hello All, I pointed my browser to http://localhost:5000 and get an error from Amazom.

root@raspberrypi:~/AlexaPi# python auth_web.py
[25/Aug/2016:18:52:50] ENGINE Listening for SIGHUP.
[25/Aug/2016:18:52:50] ENGINE Listening for SIGTERM.
[25/Aug/2016:18:52:50] ENGINE Listening for SIGUSR1.
[25/Aug/2016:18:52:50] ENGINE Bus STARTING
CherryPy Checker:
The Application mounted at '' has an empty config.

[25/Aug/2016:18:52:50] ENGINE Started monitor thread 'Autoreloader'.
[25/Aug/2016:18:52:50] ENGINE Started monitor thread '_TimeoutMonitor'.
[25/Aug/2016:18:52:50] ENGINE Serving on http://0.0.0.0:5000
[25/Aug/2016:18:52:50] ENGINE Bus STARTED
192.168.11.10 - - [25/Aug/2016:18:52:55] "GET / HTTP/1.1" 303 764 "" "Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/52.0.2743.116 Safari/537.36"

We're sorry!
An error occurred when we tried to process your request. Rest assured, we're already working on the problem and expect to resolve it shortly.

Hide Details
Error Summary
400 Bad Request
The redirect URI you provided has not been whitelisted for your application. Please add your redirect URI in the 'Allowed Return URLs' section under 'Web Settings' for your Security Profile on Amazon Developer Portal.
Request Details
scope=alexa%3Aall
redirect_uri=http%3A%2F%2F192.168.11.252%3A5000%2Fcode
response_type=code
client_id=amzn1.application-oa2-

maso27, 2016-08-25:
Have you configured the web settings at developer.amazon.com?

Allowed Origins: http://xx.xx.xx.xx:5000
Allowed Return URLS: http://xx.xx.xx.xx:5000/code

...where xx.xx.xx.xx is your IP address

shawpbx, 2016-08-26:
Yes Thank you. I didn't add the word "code" to the allow return. But after
I  obtain authorization from Amazon and saying to reboot the script keeps
running. If I stop the script and reboot and run python main.py. I get
errors.

root@raspberrypi:~/AlexaPi# python main.py
Traceback (most recent call last):
  File "main.py", line 25, in <module>
    import webrtcvad
ImportError: No module named webrtcvad

I'm using maso27 / AlexaPi
git clone --branch version1.2 https://github.com/maso27/AlexaPi.git

This is for my daughter's 4th grade science project.. I hope. :)

Thanks, David

maso27, 2016-08-26:
@shawpbx Have you run setup.sh to install everything, or did you run auth_web.py on its own?

Based on the fact that you used port 5000 above instead of 5050, (and that it worked), I'm guessing you've got files from a couple different sources.  The version of auth_web.py that's at my 1.2 branch requires port 5050 so it doesn't collide with shairport-sync.

If you haven't, run the setup.sh from that branch.  It will install the necessary packages.

And I hope your daughter learns a ton!

shawpbx, 2016-08-26:
Thanks, she is loving very minute on the RasPi.

Quick Note. I start with a clean 2016-05-27-rasbian-jessie. I login as pi
and changed root password. Then su to root. change to /root dir. I run git
clone --branch version1.2 https://github.com/maso27/AlexaPi.git then cd to
Alexa. run ./setup. I enter the Amazon questions. Then it asked me to go to
http://xxx:5050. I connect to the webpage and get the "Success! Refresh
token has been added to your creds file, you may now reboot the Pi" But the
script doesn't stop.  Should I kill the script "CTRL C" or wait??

./setup.sh
--Creating creds.py--
Enter your Device Type ID:
AlexaPi
Enter your Security Profile Description:
AlexaPi
Enter your Security Profile ID:
amzn1.application.161fdb949b4f4733axxxxxxxxxxxxxxx
Enter your Client ID:
amzn1.application-oa2-client.51c31f1xxxxxxxxxxxxxxxxxxxxxxxx
Enter your Client Secret:
a73462cbfa8344021a0f831e77033a6xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
Ready goto http://192.168.11.252:5050 or http://localhost:5050  to begin
the auth process

Thanks agian for your time.

David

maso27, 2016-08-26:
Yes, Ctrl-C to exit.  That script never stops.
(Hmmm.  Maybe a note about that would be helpful?)

Sounds like you're nearly there!  Next step: figuring out audio devices.

illperipherals, 2016-08-26:
Isn't that always the fun part? ;)

shawpbx, 2016-08-26:
She is 9y/o and has built a RasPBX phone system for her class. The kids
love playing with it. We added app-rpt and two frs radios to the system. I
know the Pi's are only 35 bucks but its adding up fast. :)

David

shawpbx, 2016-08-26:
Here is the log after a fresh install and reboot.

root@raspberrypi:~ $ cat /var/log/alexa.log
Starting Alexa in silent mode...
Traceback (most recent call last):
  File "/root/AlexaPi/main.py", line 25, in <module>
    import webrtcvad
ImportError: No module named webrtcvad

David

maso27, 2016-08-26:
Strange.  @jaydev1997 was seeing this same issue yesterday too.

Try `pip install webrtcvad` and tell me if there's a new error when you run main.py

shawpbx, 2016-08-26:
So I installed pip install webrtcvad and ran python main.py.

Successfully installed webrtcvad
Cleaning up...
root@raspberrypi:~/AlexaPi# python main.py
Traceback (most recent call last):
  File "main.py", line 27, in <module>
    from pocketsphinx.pocketsphinx import *
  File "/usr/local/lib/python2.7/dist-packages/pocketsphinx/__init__.py",
line 37, in <module>
    from pocketsphinx import *
  File
"/usr/local/lib/python2.7/dist-packages/pocketsphinx/pocketsphinx.py", line
42, in <module>
    _pocketsphinx = swig_import_helper()
  File
"/usr/local/lib/python2.7/dist-packages/pocketsphinx/pocketsphinx.py", line
38, in swig_import_helper
    _mod = imp.load_module('_pocketsphinx', fp, pathname, description)
ImportError: libpocketsphinx.so.3: cannot open shared object file: No such
file or directory

After reboot

pi@raspberrypi:~ $ cat /var/log/alexa.log
Starting Alexa in silent mode...
Traceback (most recent call last):
  File "/root/AlexaPi/main.py", line 27, in <module>
    from pocketsphinx.pocketsphinx import *
  File "/usr/local/lib/python2.7/dist-packages/pocketsphinx/__init__.py",
line 37, in <module>
    from pocketsphinx import *
  File
"/usr/local/lib/python2.7/dist-packages/pocketsphinx/pocketsphinx.py", line
42, in <module>
    _pocketsphinx = swig_import_helper()
  File
"/usr/local/lib/python2.7/dist-packages/pocketsphinx/pocketsphinx.py", line
38, in swig_import_helper
    _mod = imp.load_module('_pocketsphinx', fp, pathname, description)
ImportError: libpocketsphinx.so.3: cannot open shared object file: No such
file or directory

Thanks, David

maso27, 2016-08-26:
Huh.  I don't understand it, because I tested this on a fresh install.  But it looks like something's wrong with setup.sh.  Install packages missing.

Stand by...

maso27, 2016-08-27:
@shawpbx 
I have another idea.

Have you run `raspi-config` to expand the raspberry pi image to fill the SD card?

Type `df -h` and see if you've run out of space.

shawpbx, 2016-08-27:
Looks like I have 3.0G left.

pi@raspberrypi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.2G  3.9G  3.0G  57% /
devtmpfs        459M     0  459M   0% /dev
tmpfs           463M     0  463M   0% /dev/shm
tmpfs           463M  6.3M  457M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           463M     0  463M   0% /sys/fs/cgroup
/dev/mmcblk0p1   63M   21M   43M  33% /boot
tmpfs            93M     0   93M   0% /run/user/1000

shawpbx, 2016-08-27:
I ran ./setup and saved the output to a file (see attachment). Looks like
pocketsphinx didn't install. I installed using apt-get install
pocketsphinx. re-ran python main.py. I get a new error. So I did a reboot
and Alexa works.

root@raspberrypi:~/AlexaPi# python main.py
Checking Internet Connection...
Connection OK
Traceback (most recent call last):
  File "main.py", line 592, in <module>
    start()
  File "main.py", line 515, in start
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: Device or resource busy [plughw:1]

If you want me to do a fresh OS install and test a new version for you
please let me know. Thank you for all your help. You got us pointed in the
right direction.

Thanks, David

shairport-sync (Airplay) WILL be installed.
monitoring WILL be installed.
Hit http://archive.raspberrypi.org jessie InRelease
Get:1 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]
Hit http://archive.raspberrypi.org jessie/main armhf Packages
Get:2 http://mirrordirector.raspbian.org jessie/main armhf Packages [8,981 kB]
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Get:3 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [37.5 kB]
Get:4 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [70.3 kB]
Get:5 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1,356 B]
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Fetched 9,105 kB in 28s (324 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
git is already the newest version.
wget is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
--copying pocketsphinx--
Reading package lists...
Building dependency tree...
Reading state information...
memcached is already the newest version.
python-alsaaudio is already the newest version.
python-dev is already the newest version.
python-pip is already the newest version.
swig is already the newest version.
vlc is already the newest version.
libasound2-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Requirement already satisfied (use --upgrade to upgrade): Wave>=0.0.2 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 1))
Requirement already satisfied (use --upgrade to upgrade): python-memcached>=1.50 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 2))
Requirement already satisfied (use --upgrade to upgrade): requests>=2.4.3 in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 3))
Requirement already satisfied (use --upgrade to upgrade): wsgiref>=0.1.2 in /usr/lib/python2.7 (from -r requirements.txt (line 4))
Requirement already satisfied (use --upgrade to upgrade): CherryPy>=4.0.0 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 5))
Requirement already satisfied (use --upgrade to upgrade): python-vlc>=1.1.2 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 6))
Requirement already satisfied (use --upgrade to upgrade): pyalsaaudio>=0.8.2 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 7))
[... 184 lines cut ...]
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DSPHINXBASE_EXPORTS -DPOCKETSPHINX_EXPORTS -DSPHINX_DLL -DHAVE_CONFIG_H -Ideps/sphinxbase/include -Ideps/sphinxbase/include/sphinxbase -Ideps/sphinxbase/include/android -I/usr/include/python2.7 -c deps/sphinxbase/src/libsphinxbase/util/genrand.c -o build/temp.linux-armv7l-2.7/deps/sphinxbase/src/libsphinxbase/util/genrand.o -Wno-unused-label -Wno-strict-prototypes -Wno-parentheses -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-result -Wno-sign-compare

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DSPHINXBASE_EXPORTS -DPOCKETSPHINX_EXPORTS -DSPHINX_DLL -DHAVE_CONFIG_H -Ideps/sphinxbase/include -Ideps/sphinxbase/include/sphinxbase -Ideps/sphinxbase/include/android -I/usr/include/python2.7 -c deps/sphinxbase/src/libsphinxbase/util/f2c_lite.c -o build/temp.linux-armv7l-2.7/deps/sphinxbase/src/libsphinxbase/util/f2c_lite.o -Wno-unused-label -Wno-strict-prototypes -Wno-parentheses -Wno-unused-but-set-variable -Wno-unused-variable -Wno-unused-result -Wno-sign-compare

[... cut ...]
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DSPHINXBASE_EXPORTS -DPOCKETSPHINX_EXPORTS -DSPHINX_DLL -DHAVE_CONFIG_H -Ideps/sphinxbase/include -Ideps/sphinxbase/include/sphinxbase -Ideps/sphinxbase/include/android -I/usr/include/python2.7 -c deps/sphinxbase/src/libsphinxbase/util/cmd_ln.c -o build/temp.linux-armv7l-2.7/deps/sphinxbase/src/libsphinxbase/util/cmd_ln.o -Wno-unused-label -Wno-strict-protot

maso27, 2016-08-28:
Glad you had success!
It sounds like pocketsphinx has been moved into the Raspbian repisitory.

When I first started building my branch, I had to build sphinxbase and pocketsphinx from their guthubs, but since then it all moved into pip, and now into apt.  I'll have to make an adjustment and see how it goes.

maso27, 2016-08-30:
@shawpbx I've found the problem and fixed the repository.
The fix is to `apt-get install libpulse-dev` before the `pip install -r requirements.txt`

(I could never get the `apt-get install pocketsphinx python-pocketsphinx` approach working.  If you try this fix, you might have to run `apt-get --purge remove pocketsphinx` before doing it.)

shawpbx, 2016-08-30:
Ok I'll run a fresh OS and give it a try. Let you know later today.

David

renekliment, 2016-10-26:
Do you still have issues with this?

shawpbx, 2016-10-26:
No, Mason fixed it and should work now. My install is working fine now.

David

renekliment, 2016-10-26:
Great! Closing.

closed 2016-10-26 by renekliment