alexa-pi/AlexaPiDEPRECATED #116: response 204 again

fenyvesi, 2016-08-07:
I use V1.1.  It worked a week ago. Yesterday  I tried again after installing some other sws(VNC, mosquitto).
It didn't work. I installed AlexaPi again.
The result is the same:
I get always error 204, no Alexa sound answer.
the debug print:

             root@raspberrypi:/home/pi# python /root/AlexaPi/main.py
             Checking Internet Connection...
            Connection OK
            Play_Audio Request for: /root/AlexaPi/hello.mp3
            Player State: State.Opening
            Player State: State.Playing
            Player State: State.Ended
            Ready to Record.
            Recording...
            Recording Finished.
            Sending Speech Request...
            Processing Request Response...
            Request Response is null (This is OKAY!)
            Ready to Record.

I can play the recorded sound correctly this way, so it seems to be OK:
            aplay -vv -F 500 -r 16000 -c 1 -f S16_LE -D plughw:1 /root/AlexaPi/recording.wav

Somebody already sent the same issue and said it was a permission problem. I have no idea what it is.

Do you have any suggession?

Thank you.

George

mochman, 2016-08-07:
I have received the same error.  I created a wav file on another computer with a clearer mic using S16_LE 16k mono audio and forced main.py to send that file instead.  I get back the 204 response as well.

fenyvesi, 2016-08-07:
Thank you. So the error is in the recording.

maso27, 2016-08-07:
This is a bigger issue, within Amazon itself.

https://forums.developer.amazon.com/questions/33371/204-response-1.html

I've seen a couple reports that new accounts are working but old ones
aren't, but I don't think anyone knows what is going on right now. My Alexa
installation quit working last Thursday, and as of right now is still
returning null responses.

fenyvesi, 2016-08-07:
I started a virgin RasPi installation but I interrupt it and wait for Amazon. Maybe nothing special is on my side.

Thank you for the information.

mochman, 2016-08-07:
@maso27 Yeah, that's what it seems like the issue is.  Maybe they are doing away with APIv1 and moving fully to v2?

illperipherals, 2016-08-07:
I *really* don't believe that is the issue.

maso27, 2016-08-08:
I just checked mine, and Alexa is back!

Anyone else having the same success?

fenyvesi, 2016-08-08:
I am OK again.

renekliment, 2016-10-26:
Seems like fixed -> closing. Reopen if needed.

closed 2016-10-26 by renekliment