alexa-pi/AlexaPiDEPRECATED #117: Alexa not starting on boot... how to change?

andygbennett, 2016-08-08:
Alexa is not starting on boot. In /var/log/alexa.log, it says:

"Starting Alexa...
python: can't open file '/root/AlexaPi/main.py': [Errno 2] No such file or directory"

My AlexaPi directory is located at /home/pi/AlexaPi

How can I change the file path to /home/pi/AlexaPi/main.py so that it starts on boot??

Thanks :)

maso27, 2016-08-08:
I can think of two ways.

1.) moving /home/pi/AlexaPi to /root/AlexaPi

2.) editing /etc/init.d/AlexaPi

This project was designed to be run as root, in the /root directory.  So I'm not sure what permissions issues may crop up after moving it.  But see what you can see!

sammachin, 2016-09-01:
Need to improve install instructions to avoid this problem in future.

closed 2016-09-01 by sammachin