alexa-pi/AlexaPiDEPRECATED #118: ALSA lib PCM_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave

Beaverboy, 2016-08-09:
My alexa works up until the point when it tries to play back the response. It then issues the above error as well as 
alsa audio output error: cannot open ALSA device "default": no such file or directory 
core audio output error: Audio output failed
core audio output error: The audio device "default" could not be used : no such file or directory.
core audio output error: module not functional 
core decoder error: failed to create audio output. 

I can use aplayer to play a .wav file with no errors. I'm not really sure what to do from here.

Beaverboy, 2016-08-11:
I am using the headphone jack on a raspberry pi 3. I have been stuck at this point for days :'( would really appreciate some input, or some other things to try.
ive tried changing the alsa.conf file in order to resolve this issue, but havent had any luck.

maso27, 2016-08-11:
I'm not sure which version of AlexaPi you're running, but you might try changing any line in main.py that uses "vlc.Instance" to the following:

`vlc.Instance('--aout=alsa', '--alsa-audio-device=plughw:0,0')`

Beaverboy, 2016-08-11:
I fixed it for the time being. Ill post some instructions about what I had to do to fix it, seems my issue was related to the different asoundrc configurations ( one for root ~/.asoundrc and one for user pi /etc/asound.conf ), it seems I only had to specify what the hardware care values were in both locations.

closed 2016-09-01 by sammachin