alexa-pi/AlexaPiDEPRECATED #119: Traceback (most recent call last):   File "main.py", line 25, in <module>     import webrtcvad ImportError: No module named webrtcvad

jaydev1997, 2016-08-24:
Traceback (most recent call last):
  File "main.py", line 25, in <module>
    import webrtcvad
ImportError: No module named webrtcvad

maso27, 2016-08-24:
I'm guessing this is for my 1.2 version, considering that's where there's a main.py where line 25 says `import webrtcvad`.

Somehow you've missed the installation of webrtcvad.  Try typing `pip install webrtcvad` from the command terminal and see if that helps.

jaydev1997, 2016-08-24:
It works and another error import pocketsphinx.pocketsphinx

jaydev1997, 2016-08-24:
i tried to install :  pip install pocketsphinx

but i got this error...   -- http://hastebin.com/utazadizar.tex

maso27, 2016-08-24:
Did you install AlexaPi by running setup.sh?

1.) Login as root
2.) Place the directory labeled "AlexaPi" in /root
3.) Run `/root/AlexaPi/setup.sh` and follow the instructions.

These things should be taken care of if you've already done that.

illperipherals, 2016-08-24:
Have you cats been looking at silence detection?

maso27, 2016-08-24:
Yeah.  It works quite well on my 1.2 branch.

illperipherals, 2016-08-24:
I am using snowboy at this point, but wanting to pick a lot of your button stuff. What is the best upstream there?

illperipherals, 2016-08-24:
Also, maso27, here is a pretend trophy for just being a badass carrying this project on.
It may not be heavy, but it is badass.

jaydev1997, 2016-08-24:
@illperipherals  Can you Tell Me how to set Alexa with snowboy..?

maso27, 2016-08-24:
@illperipherals Thanks, man.  My family uses my AlexaPi daily, so it's fun to keep thinking about it.
As to the button stuff--1.2 is good, but I took bdatdo0601's snowboy fork and put some of my things into it too.

I've been wanting to get it into github but I just started a new job so my time is limited lately.
Check out this main.py.  It's got vad, button, and follow-up question listening in it.
I had to modify snowboydecoder.py to get it to work though (search in the attached for "main.button.pressed" and you'll see it).

Hopefully you can do something with that.
[snowboybits.zip](https://github.com/sammachin/AlexaPi/files/435234/snowboybits.zip)

maso27, 2016-08-24:
@jaydev1997 It took me some work to get snowboy working.  I took notes for what I did, and maybe they'll be helpful:

(note: login as pi instead of root for this)

Git clone https://github.com/bdatdo0601/AlexaPi 
apt-get update 
apt-get install swig3.0 python-pyaudio python3-pyaudio sox 
pip install pyaudio 
apt-get install libatlas-base-dev 
apt-get install python-pygame 
Run as pi with "python start.py Alexa.pmdl"

illperipherals, 2016-08-24:
@maso27, nice! Give me a bit to dig in. Break a leg on the new job.

@jaydev1997, I have some notes that I have whipped myself into making. What issues are you having? I assume they are probably audio...

Can you do just a general lsusb, aplay -L, arecord -L dump?

illperipherals, 2016-08-24:
I was able to resolve a couple issues here:
https://github.com/Kitt-AI/snowboy/issues/9#issuecomment-241577381

jaydev1997, 2016-08-24:
@illperipherals 
lsusb         --> http://hastebin.com/utaneyusoh.sm
aplay -L     --> http://hastebin.com/yiyiqipive.sm
arecord -L dump     --> http://hastebin.com/gusewuqeha.avrasm

jaydev1997, 2016-08-24:
@maso27 
i still got this error

root@pi:/home/pi/AlexaPi# python main.py
Traceback (most recent call last):
  File "main.py", line 27, in <module>
    from pocketsphinx.pocketsphinx import *
ImportError: No module named pocketsphinx.pocketsphinx

flooie, 2016-08-24:
You have the wrong repo...?  maybe

jaydev1997, 2016-08-24:
i already done the setup also the localhost:5050 setup.. then i rebooted.

maso27, 2016-08-26:
@jaydev1997 If you're still messing with this--I've discovered something wrong with setup.sh.  Stay tuned...

pagespec1, 2016-08-28:
I had the same webrtcavd error. Running the pip install webrtcvad got me past the error, but thei I also received the pocketsphinx error as @jaydev1997. The error message seemed to indicate that pocketsphinx could not install because pulseaudio is missing or at least "pulse/pulseaudio.h" is not where it should be. I tried installing pulseaudio, but it does not appear to have helped. I chose not to install airplay at the beginning of the setup. I don't know if that has any bearing on this.

pagespec1, 2016-08-28:
@jaydev1997 If you do an `apt-get install libpulse-dev` it will fix the issues with pocketsphinx. After installing that library and running the `pip install -r requirements.txt` my Alexa client started in silent mode successfully.

maso27, 2016-08-30:
@pagespec1 Yes!  That fixes it!
I hadn't seen your posts, and just went through discovering the same thing.

In a nutshell: "pip install pocketsphinx" installs sphinxbase first, in order to build pocketsphinx.  Sphinxbase has a (new?) dependency for libpulse-dev, so it errors out if it doesn't have that.  Webrtcvad just doesn't get installed.

I've fixed the repo so that should no longer be an issue in version1.2 of my branch.

(as an aside: "apt-get install pocketsphinx python-pocketsphinx" does NOT work.  Even after fixing the includes, it can't find "Decoder.default_config()")

renekliment, 2016-10-26:
Do you still have issues with this?

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment