alexa-pi/AlexaPiDEPRECATED #120: Is TuneIn Radio Working?

GadgetReactor, 2016-09-02:
Hi I just loaded 1.1 and was trying to play a TuneIn Radio channel. Did they change the format? I am getting a NoneType error. Thank you.

sammachin, 2016-09-02:
Whats the name of the station, from what I've found their format varies so I've had to put workarounds in for many of them. you could try `play bbc radio one on tune in` as a test command

123icebuggy, 2016-09-03:
When I do that, it says `BBC Radio on Tune in`, then does nothing...

GadgetReactor, 2016-09-06:
Thanks. Looks like the tunein playlist is a mess. I was able to get BBC Radio One to play. @123icebuggy I had to make one change though. - to stop the audioplaying check. 

`
		while(GPIO.input(button)==0): # we keep recording while the button is pressed
`
`
			if audioplaying: p.stop() # moved this down
`

123icebuggy, 2016-09-06:
What about Triple J?

Thanks anyway, I was able to get it to work, but your way seems faster

closed 2016-10-19 by GadgetReactor