alexa-pi/AlexaPiDEPRECATED #122: You alexa device is currently offline

123icebuggy, 2016-09-03:
I want to control Alexa from my phone, but it thinks it is offline...

Sorry to clog this issues page, I really do like what you've done.

tangowhisky37, 2016-09-12:
Sam - I have a similar issue. My Pi is definitely logged in and connected, I am able to press the button and talk to Alexa. However, the web interface (alexa.amazon.com) says that i am offline.

Am sure there's something different in the way we've setup the pi and use the python code you've developed. When using the reference java application the web interface (alexa.amazon.com) does show the device as online.

Able to help us out here with what changes might be required?

maso27, 2016-09-12:
AlexaPi always reports as being "offline."  It's not an issue.  It's a function of the way the system works, only sending audio files to Amazon when there's something worth sending.

Other systems maintain an always-on server that is connected to Amazon.  They register as being "online."

sammachin, 2016-09-25:
Yeah this is mostly due to us still using the V1 Alexa API, once we get onto the V20160217 api and have a persistent HTTP/2 connection things like pushing from the app will become possible, might not work right away though!

123icebuggy, 2016-09-25:
Would an ETA on that be available? Also could you look at this https://github.com/sammachin/AlexaPi/issues/123 issue

renekliment, 2016-10-26:
We're very probably gonna do this on the [new repo](https://github.com/alexa-pi/AlexaPi) only. You can check it out in the meantime.

closed 2016-10-26 by renekliment