alexa-pi/AlexaPiDEPRECATED #123: Error with certain TuneIn stations.

123icebuggy, 2016-09-03:
I am trying to play Triple J on TuneIn 

But I get an error of 
`"Sorry, something went wrong"`

However BBC Radio 1 works just fine.

Any ideas?

123icebuggy, 2016-09-03:
It seems I cannot get any Australian stations working

123icebuggy, 2016-09-03:
I also get a lot of 'I'm not sure what went wrong'

123icebuggy, 2016-09-03:
Some stations give me this error:

Exception in thread Thread-9:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "main.py", line 270, in play_audio
    m = i.media_new(file)
  File "build/bdist.linux-armv7l/egg/vlc.py", line 1608, in media_new
    if ':' in mrl and mrl.index(':') > 1:
TypeError: argument of type 'NoneType' is not iterable

^CTraceback (most recent call last):
  File "main.py", line 410, in <module>
    start()

123icebuggy, 2016-09-04:
Now BBC won't work either, saying `Sorry, I can't reach TuneIn`

sammachin, 2016-09-25:
The API switch is something we need to do but I don't have an ETA at the moment, we're currently in the processes of moving the project and codebase onto a more collaborative effort as a standalone project which should make it easier for more people to contribute and hopefully get more stuff working faster.

The TuneIn stuff is improved in the new project but I can't get "play triple j on tune in to work on there either" 

However there seems to be bigger issues with TuneIn from Alexa as I am getting the same errors using my TAP or Echo, Amazon are doing a lot of work to ready for the UK launch this week so I suspect things like radio stations especially outside the US are in a state of flux. I'd suggest sitting tight on TuneIn issues for a few weeks until it settles down. Hopefully we'll have the new project ready then too.

123icebuggy, 2016-09-25:
Ok thanks, I really want it because at the moment I am using a voice activated script to listen to that radio, problem is, when I use AlexaPi, it throws alsa errors saying the device is already in use, any way around this?

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment