alexa-pi/AlexaPiDEPRECATED #124: ImportError: No module named alsaaudio

jaydev1997, 2016-09-03:
File "main.py", line 7, in <module>
    import alsaaudio
ImportError: No module named alsaaudio

123icebuggy, 2016-09-03:
Please confirm you are using `python` and not python3`

matthewtholt, 2016-10-09:
I'm getting the same error. definitely using python. @jaydev1997  did you get this sorted?

jaydev1997, 2016-10-09:
@MRedRaider Yes i tried configured Audio Channels in Alsa and it sorted out... first confirm that you have Alsa with ```alsamixer```

closed 2016-10-09 by jaydev1997

coach0742, 2016-10-09:
So I'm having this same issue and I can't figure out what is wrong. I have already modified main.py with the device name outputted by arecord -L, I have tested the mic with arecord and am getting input just fine. But every time I try to start up main.py, I get this error.

jaydev1997, 2016-10-10:
which python version do you have? @coach0742

MRedRaider, 2016-10-12:
I'm going to admit my mistake here, I'm working on a CHIP, followed that repo and when I googled this error got here and didn't think twice since I also have this running on a RPI sitting beside it. Sorry!

jaydev1997, 2016-10-12:
@MRedRaider Oh..thats ok mate...