alexa-pi/AlexaPiDEPRECATED #128: Objc version

flooie, 2016-09-09:
I wasn't sure where to put this question, but I built an objective-c Mac version of this tool - and was wondering if anyone is interested in working on the project with me.  Right now it takes text and voice inputs but is a bit buggy with pause and resume.  Just wanted to gauge interest before I put it up.

[image]

maso27, 2016-09-12:
Sounds interesting--how are you doing the text?  TTS, then sending a wav file?

Sadly, I'm not available to work on it--too many time obligations of late.  But here's hoping you have luck with it!

flooie, 2016-09-12:
Spot on @maso27 - Cocoa frameworks allow for silent text to speech file - which gets covered to wav and sent off.  I posted the framework minus my amazon auth keys. if you wanted it for your own use

renekliment, 2016-09-12:
Unfortunately, I don't have a Mac to even test this. Best of luck with it though.
The new AlexaPi should however support OS X when it's ready. Let's see where the future takes us :-)

flooie, 2016-09-12:
@renekliment the current alexapi already works on my Mac - I rewrote it in objc because I wanted an interface.

renekliment, 2016-09-18:
@flooie Have you tried doing the GUI in python for the current project? What's the advantage of doing it in pure ObjectiveC? (speed?)
Also, I'm curious what is your use case that you want a GUI. Would you share?

flooie, 2016-09-18:
@renekliment I did a number of things - and I released the Mac Echo (Mecho) on my github account.  I did build a version that ran the python version with objc before I decided to build it purely in objc.  It worked fine enough.  I did not build a python gui for the Mac version just the hybrid.

closed 2017-03-01 by renekliment