alexa-pi/AlexaPiDEPRECATED #129: ./setup.sh

Grumpyguard, 2016-09-11:
I copied and pasted the git clone statement and when I type ./setup.sh I get the following error message  -bash: ./setup.sh: No such file or directory.  What did I do wrong?  When I attempt to reissue the git clone statement I get the following message fatal: destination path 'AlexaPi' already exists and is not an empty directory.  Any help you can give me is greatly appreciated.

maso27, 2016-09-11:
You need to enter the directory that contains _setup.sh_ before trying to execute it.
`cd AlexaPi <Enter>` before the `./setup.sh` command.

You are logged in as root, starting from the /root directory, right?

Grumpyguard, 2016-09-11:
Thank you.  I did not know I had to be logged in as root.

fireshipmate, 2016-09-14:
Hello all, 
my question is why I have to install the program in the root directory?
Would it not be possible to install somewhere else in the file structure using the "sudo" option?
Is it intended to start the program any time manually or would it start automatically when I power on the RaPi.?

renekliment, 2016-09-14:
Right now, the `/root` directory is hard-coded in the setup script and in the init scripts. You can use another path, if you change that.
`sudo` runs commands as the user root, it has nothing to do with paths.
This behavior is gonna change in a new version. You can either change that manually, or just wait.

You can do both. If you install it using the setup script, it's gonna start automatically upon system boot.

illperipherals, 2016-09-18:
Pulling a clean build still requires "root" to be edited to "opt" in two places. Both show up in the log. Verified this at an Alexa hackathon yesterday. Should this be rolled into an active issue, or a new PR?

renekliment, 2016-10-26:
You can switch over to the new upstream (https://github.com/alexa-pi/AlexaPi) and you can install AlexaPi outside of `/root` (the new default is `/opt`) and you can even run it under a non-root user for security.

Please let us know if you still have issues or we can close this. Thanks!

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment