alexa-pi/AlexaPiDEPRECATED #130: ImportError: No module named pocketsphinx

link2patel, 2016-09-13:
i am getting below error..how to solve it?
pi@raspberrypi:~/AlexaPi $ sudo python main.py
Traceback (most recent call last):
  File "main.py", line 27, in <module>
    from pocketsphinx.pocketsphinx import *
ImportError: No module named pocketsphinx
  
i tried pip install and i got this

 pi@raspberrypi:~/AlexaPi $ pip install -r requirements.txt
Requirement already satisfied (use --upgrade to upgrade): Wave>=0.0.2 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 1))
Requirement already satisfied (use --upgrade to upgrade): python-memcached>=1.50 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 2))
Requirement already satisfied (use --upgrade to upgrade): requests>=2.4.3 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 3))
Requirement already satisfied (use --upgrade to upgrade): wsgiref>=0.1.2 in /usr/lib/python2.7 (from -r requirements.txt (line 4))
Requirement already satisfied (use --upgrade to upgrade): CherryPy>=4.0.0 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 5))
Requirement already satisfied (use --upgrade to upgrade): python-vlc>=1.1.2 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 6))
Requirement already satisfied (use --upgrade to upgrade): pyalsaaudio>=0.8.2 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 7))
Requirement already satisfied (use --upgrade to upgrade): py-getch>=1.0.1 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 8))
Requirement already satisfied (use --upgrade to upgrade): pocketsphinx>=0.1.0 in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 9))
Requirement already satisfied (use --upgrade to upgrade): webrtcvad>=2.0.7 in /usr/local/lib/python2.7/dist-packages (from -r requirements.txt (line 10))
Requirement already satisfied (use --upgrade to upgrade): six>=1.4.0 in /usr/local/lib/python2.7/dist-packages/six-1.10.0-py2.7.egg (from python-memcached>=1.50->-r requirements.txt (line 2))

and while running setup.sh i got this yellow text

 Found existing installation: pyalsaaudio 0.7
    DEPRECATION: Uninstalling a distutils installed project (pyalsaaudio) has been deprecated and will be removed in a future version. This is due to the fact that uninstalling a distutils project will only partially uninstall the project.
    Uninstalling pyalsaaudio-0.7:
      Successfully uninstalled pyalsaaudio-0.7

maso27, 2016-09-13:
Looks like you're trying to run as _pi_ instead of _root_.
And based on the fact that it wants pocketsphinx, you're not running the one version (maso27/snowboy) that is designed to be run as _pi_.

Which version are you using?  Most likely, modules are looking for items in _/root_, while your whole install is in _/home/pi_.

Do you have a _pocketsphinx_ directory, either in _/root_ or _/home/pi_? 
 _main.py_ is looking for that.

You can potentially perform a quick fix by moving _pocketsphinx_ into the directory that it's seeking.  Might take more than that though.

renekliment, 2016-10-26:
Seems like you're running Mason's `version1.2` branch. Use the new upstream instead: https://github.com/alexa-pi/AlexaPi It is based on that one, but has enhancements. Follow the installation guide.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment