alexa-pi/AlexaPiDEPRECATED #131: Issues playing tunein stations

interbiznw, 2016-09-15:
Not sure why it won't play most stations but heres some output:

`Ready to Record.
Recording...
Recording Finished.
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Music#ACRI#DeviceTTSRendererV4_8238283f-d812-452c-851e-30e35ef2dedb", "audioContent": "cid:DeviceTTSRendererV4_8238283f-d812-452c-851e-30e35ef2dedb_1326322354"}}, {"namespace": "AudioPlayer", "name": "play", "payload": {"playBehavior": "REPLACE_PREVIOUS", "audioItem": {"streams": [{"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 900000}, "streamUrl": "https://opml.radiotime.com/Tune.ashx?serial=AEGBPNIX5KOMRPXWY6KUIOMJMHCA&formats=aac,mp3&partnerId=4JqugguZ&id=s35880", "offsetInMilliseconds": 0, "expiryTime": "2016-09-15T20:29:57+0000", "streamId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#4e12fc31-da8e-4990-94a2-f96ec4dce4ce:1", "progressReportRequired": true}], "audioItemId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#4e12fc31-da8e-4990-94a2-f96ec4dce4ce:1"}, "navigationToken": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#4e12fc31-da8e-4990-94a2-f96ec4dce4ce:1"}}]}, "messageHeader": {}}
Play_Audio Request for: /home/pi/AlexaPi/tmpcontent/DeviceTTSRendererV4_8238283f-d812-452c-851e-30e35ef2dedb_1326322354.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Play_Audio Request for: https://opml.radiotime.com/Tune.ashx?serial=AEGBPNIX5KOMRPXWY6KUIOMJMHCA&formats=aac,mp3&partnerId=4JqugguZ&id=s35880
Ready to Record.
Player State: State.Opening
[ac8405d8] es demux error: cannot peek
[ac8405d8] es demux error: cannot peek
Player State: State.Playing
Sending Playback Progress Report Request...
Player State: State.Ended
Sending Playback Progress Report Request...
Sending GetNextItem Request...
Playback Progress Report was Successful!
Playback Progress Report was Successful!
Processing Request Response...
JSON String Returned: {"messageBody": {"audioItem": {"streams": [{"progressReport": {"progressReportDelayInMilliseconds": 0, "progressReportIntervalInMilliseconds": 900000}, "streamUrl": "https://opml.radiotime.com/Tune.ashx?serial=AEGBPNIX5KOMRPXWY6KUIOMJMHCA&formats=aac,mp3&partnerId=4JqugguZ&id=s35880", "offsetInMilliseconds": 0, "expiryTime": "2016-09-15T20:30:05+0000", "streamId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#4e12fc31-da8e-4990-94a2-f96ec4dce4ce:1", "progressReportRequired": true}], "audioItemId": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#4e12fc31-da8e-4990-94a2-f96ec4dce4ce:1"}, "navigationToken": "amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#4e12fc31-da8e-4990-94a2-f96ec4dce4ce:1"}, "messageHeader": {}}
Play_Audio Request for: https://opml.radiotime.com/Tune.ashx?serial=AEGBPNIX5KOMRPXWY6KUIOMJMHCA&formats=aac,mp3&partnerId=4JqugguZ&id=s35880
Player State: State.Opening
[b01ab410] es demux error: cannot peek
[b01ab410] es demux error: cannot peek
Player State: State.Playing
Sending Playback Progress Report Request...
Player State: State.Ended
`

sammachin, 2016-09-25:
TuneIn seems to be very broken right now I'm even getting errors on my Amazon hardware (TAP and Echo) I think they are moving stuff around ready for the UK/DE launch of the Echo this week. 

I'd sit tight and I'll report back when its working on the Amazon HW then we can look into AlexaPi issues.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment