alexa-pi/AlexaPiDEPRECATED #133: After Alexa starts, alsaaudio.ALSAAudioError: Input/output error

sikkz, 2016-09-26:
Hello,
I receive the "Hello" from Alexa and the green LED flashes 3 times.
But after pushing (and holding) the button, I only receive this output:

Checking Internet Connection
Connection OK
Traceback (most recent call last):
  File "main.py", line 156, in <module>
    start()
  File "main.py", line 130, in start
    l, data = inp.read()
alsaaudio.ALSAAudioError: Input/output error

I googled the error, and checked the Issues page here, but can't find a solution.
I am using a Raspberry Pi 3 and a Webcam with integrated microphone.
If anybody has a solution, please try to explain it well, because I am just starting to learn programming.

Thanks for your help in advance!

Tiqni, 2016-10-03:
You are not calling correctly the microphone, check the solution here it may work for you:
https://github.com/sammachin/AlexaPi/issues/135

sikkz, 2016-10-06:
First, thank you for your reply!
I ran the arecord -L command. Because of the results, I changed:
"plughw:1 to plughw:Camera=Device,DEV=0"
But still I only receive the "hello" and three green flashes.
Anymore ideas?

## Heres the arecord -L output:

```
pi@raspberrypi:~/Downloads/AlexaPi $ arecord -L                                 null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
sysdefault:CARD=Camera
    USB Video Camera, USB Audio
    Default Audio Device
front:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    Front speakers
surround21:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    Direct sample mixing device
dsnoop:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    Direct sample snooping device
hw:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Camera,DEV=0
    USB Video Camera, USB Audio
    Hardware device with all software conversions
```

## And heres the main.py's content:

```
#! /usr/bin/env python

import os
import random
import time
import RPi.GPIO as GPIO
import alsaaudio
import wave
import random
from creds import *
import requests
import json
import re
from memcache import Client

#Settings
button = 18 #GPIO Pin with button connected
lights = [24, 25] # GPIO Pins with LED's conneted
device = "plughw:CARD=Camera,DEV=0" # Name of your microphone/soundcard in arec$

#Setup
recorded = False
servers = ["127.0.0.1:11211"]
mc = Client(servers, debug=1)
path = os.path.realpath(__file__).rstrip(os.path.basename(__file__))

def internet_on():
    print "Checking Internet Connection"
    try:
        r =requests.get('https://api.amazon.com/auth/o2/token')
        print "Connection OK"
        return True
    except:
        print "Connection Failed"
        return False

def gettoken():
        token = mc.get("access_token")
        refresh = refresh_token
        if token:
                return token
        elif refresh:
```

renekliment, 2016-10-26:
Can you try the new upstream (https://github.com/alexa-pi/AlexaPi)? The installation over there is more robust and it might be therefore easier to get it to work. Just follow the installation guide. Thanks!

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment