alexa-pi/AlexaPiDEPRECATED #134: Issue when getting to the login screen for AlexaPi

TheFonix, 2016-09-27:
I'm having the same issue, but i already have the device name set to my_device, as soon as i hit the ok button to get the key from the AlexaPi it shows up with

We're sorry!
An error occurred when we tried to process your request

Any ideas? Seems nobody else is having this now since the device name fixed it, I've tried moving the directory I launch it from, I've tried flashing and putting a new install on rasp on, I've also tried a new amazon account and nothing seems to work. The allowed URL's are correct for sure, I've got http://localhost:5000 and http://localhost:5000/code as a return so not sure what else it could be.

Tiqni, 2016-10-03:
Are you doing this directly in the Raspberry Pi browser or by SSH?

TheFonix, 2016-10-03:
Sorry, that's through the web browser, on both my computer and then the browser inside of the raspberry pi, I doubled check the allowed url's too and they were correct, I have both local host in there and the IP address of the pi so I could access it remotely and I get past all of the login part but as soon as I click ok for it to generate the secret key I get the error message (We're sorry etc...)

Tiqni, 2016-10-03:
Try to create a new device, and in the Web Settings put again:

**Allowed Origins**   
http://localhost:5000
http://raspberrypi:5000

**Allowed Return URLs**
http://localhost:5000/code
http://raspberrypi:5000/code

These addresses are working for me.

TheFonix, 2016-10-03:
Tried that, no luck

maso27, 2016-10-03:
Here's what's required to generate the refresh token:

1.) _creds.py_ must have correct values for
- ProductID
- Security_Profile_Description
- Security_Profile_ID
- Client_ID
- Client_Secret

...available under _General_ at developer.amazon.com

2.) _Web Settings_ at developer.amazon.com must be set correctly
- Allowed Origins: http://xxx.xxx.xxx.xxx:5000 where the x's are your IP address
           (This will be 5050 instead of 5000 if you're using versions that have shairport-sync)
- Allowed Return URLs: http://xxx.xxx.xxx.xxx:5000/code
           (again, might be 5050)

If you login via your Raspi browser, type the IP address into the browser, not _localhost_.  If you would rather type _localhost_ into the browser, replace the x's above with _localhost_.  This only works from the browser within the raspberry pi.

It sounds like folks have had mixed success using _raspberrypi.local_ instead of the IP address, but again, you should type into the browser what you have entered into your _Web Settings_.

3.) run _auth_web.py_ as root, then go to port 5000 of your ip address from a browser on the local network.

I can't think of anything else that's necessary, and when I've had issues, I have made some mistake within one of the three items listed above.

TheFonix, 2016-10-05:
Ok, thanks! I'll try it out soon

renekliment, 2016-10-26:
Does it work for you yet?

illperipherals, 2016-10-26:
I have a little bash script that comes in handy. I thought I had posted it up here, but I can dig it up.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment