alexa-pi/AlexaPiDEPRECATED #135: got problem with alsaaudio... please help me dude...  stuck here

balajitk7, 2016-09-27:
"""""Traceback (most recent call last):
  File "main.py", line 592, in <module>
    start()
  File "main.py", line 515, in start
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: Device or resource busy [plughw:1]""""
-------got this error while am running my file main.py-----

----- got the below for arecord -L---------
""""root@raspberrypi:/home/pi/AlexaPi# arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions""""
--- what I want to do in main.py > settings ----

please help me guys, am very passionate about this, and am new to this project, i want to complete this ...
am a lecturer i want to give demo to my students pls reply me
--- areplay -l--
"""""
root@raspberrypi:/home/pi/AlexaPi# arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0   """""

factulas, 2016-09-27:
In main.py change plughw:1 to plughw:CARD=Device,DEV=0

balajitk7, 2016-09-28:
thanks for responding , still problem remains as it is...
.
.
.
Traceback (most recent call last):
File "main.py", line 592, in 
start()
File "main.py", line 515, in start
inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: Device or resource busy [plughw:CARD=Device,DEV=0]

balajitk7, 2016-09-28:
thank you so much it works...
thanks a lot
thank you so much

souto93, 2016-10-09:
Hi , I still have the same problem even after chaging that.
Don't know what more to do , already spent so many hours trying to fix this....

balajitk7, 2016-10-10:
Type arecord - L
Then go for "plughw" line 
Copy entire first line, mine is "plughw:CARD=Device,DEV=0".  Copy ur line in main. Py settings

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment