alexa-pi/AlexaPiDEPRECATED #136: alsaaudio.ALSAAudioError: Device or resource busy

123icebuggy, 2016-10-05:
```
Recording...
Traceback (most recent call last):
  File "main.py", line 410, in <module>
    start()
  File "main.py", line 365, in start
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, device)
alsaaudio.ALSAAudioError: Device or resource busy

```

I know why this is happening, I am using my mic with another script.

Is there a way to use it with both?

renekliment, 2016-10-26:
If you have multiple sound cards, you can set the input device in main.py to another.
Another possibility would be to use PulseAudio, but I'm not sure it's tested.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment