alexa-pi/AlexaPiDEPRECATED #138: AVS Adds Hands-Free Capabilities with API Update

interbiznw, 2016-10-06:
Great news!

[image]

123icebuggy, 2016-10-07:
Well, it is good news, but not current for AlexaPi.

If I remember correctly, there are two API versions, 1 and 2. Sammachin is working to bring API 2, because it would GREATLY improve the AlexaPi.

So let's see how it goes!

interbiznw, 2016-10-07:
with what amazon released It looks like voice activation is done through a 3rd party app (https://snowboy.kitt.ai/)

renekliment, 2016-10-26:
As @123icebuggy pointed out it's good news, but currently doesn't concern AlexaPi at all.

If you want voice activation, checkout the new upstream for this project: https://github.com/alexa-pi/AlexaPi 
Until we add snowboy/pyaudio support there, you can check out Mason's repo: https://github.com/maso27/AlexaPi/tree/snowboy

closed 2016-10-26 by renekliment