alexa-pi/AlexaPiDEPRECATED #142: alsaaudio.ALSAAudioError: Device or resource busy [plughw:CARD=Device,DEV=0]

souto93, 2016-10-09:


souto93, 2016-10-09:
I'm connecting via USB a headhphones with microphone, in the same USB.
The model is : Trust GTX 340

Ierlandfan, 2016-10-11:
I-ve setup the audio (Bluetooth JBL GO) through this 
https://bbs.nextthing.co/t/guide-to-connecting-to-a-bluetooth-speaker/4684/2
Works for the Raspberry Pi (2) at least. For now I have a PS3 Eye as microphone and a Bluetooth speaker as output device. Works better than HDMI.

jtotham, 2016-10-12:
I had this problem until I turned off the graphical desktop, with my Pi configured to boot just to the console (sudo raspi-config). Might not be practical for all but if you're building a headless voice activated system then having the GUI loaded is just a waste of resources.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment