alexa-pi/AlexaPiDEPRECATED #144: Nothing

scargill, 2016-10-17:
I have an up to date pi1 with mpg123 etc.... and latgest node red etc..

I ran the script. told me to reboot.  I ran main.py -  it says "HELLO" and terminal says connection OK.

I press the button - the red light comes on - then flashes 3 times - no sound comes out of the system.

Am I missing a stage? Not seeing any error messages on the (root) terminal...

illperipherals, 2016-10-19:
Can you run it with the "-d" switch and also check your log file?
When you run aplay/arecord -L, does everything look in shape?
Have you verified with another program that your audio is working (sox is a good tool)?

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment