alexa-pi/AlexaPiDEPRECATED #146: Success installation but run into error

fboulegue, 2016-10-20:
Hi i could install all the stuff via SSH, then it did ask me to authorize, that's what i did. So i did close the script after with "ctrl + c" script ended but nothing happened.
So i did try to run the

**python main.py**
```
Traceback (most recent call last):
  File "main.py", line 84, in <module>
    decoder = Decoder(config)
  File "/usr/local/lib/python2.7/dist-packages/pocketsphinx/pocketsphinx.py", line 228, in __init__
    this = _pocketsphinx.new_Decoder(*args)
RuntimeError: new_Decoder returned -1
```

So i also did try to run

** sudo ./monitorAlexa.sh**
but the log show

```
Starting Alexa in silent mode...
python: can't open file '/root/AlexaPi/main.py': [Errno 2] No such file or directory
```

fboulegue, 2016-10-20:
let me add i use @maso27 fork

renekliment, 2016-10-20:
If you're not using Mason's `snowboy` branch, there is no reason for using his version1.1/1.2 branch. Use the new project instead - https://github.com/alexa-pi/AlexaPi It's based on that a has a lot of enhancements (with more coming).

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment