alexa-pi/AlexaPiDEPRECATED #148: Outputting Text

Simon-Chase, 2016-11-11:
How could I make Alexa output text to be displayed in an application?

renekliment, 2016-11-11:
What exactly do you mean? You want the transcription of your voice?

Simon-Chase, 2016-11-11:
I'd like it to output my question and it's answer.

renekliment, 2016-11-11:
AFAIK it's not possible (someone please correct me if I'm wrong).

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment