alexa-pi/AlexaPiDEPRECATED #149: Spotify/Connect Bluetooth

SilenceDev, 2016-11-23:
Anyone able to connect to spotify or any other music gateway? Or connecting bluetooth?

GamefreakSenpai, 2016-12-08:
Amazon didnt had music streaming authorized for non-Amazon echo's. They did in the last update so maybe in the next AlexaPi update there will be those capabilities aswell as http/2 streaming 👍

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment