alexa-pi/AlexaPiDEPRECATED #152: setup fails to install cherrypi / requests / memcache

braddo99, 2016-11-27:
Hello, when running the setup.sh using sudo su (root) the installer mostly completes without error except the final message that cherrypi is not found.

Then when I reboot and navigate to myIP:5000 I just get "site can't be reached"

Looking at /var/log.alexa.log there is a message that module requests not found.

I did sudo apt-get install python-requests and reboot again.

Now /var/log says no module named memcache ...  lots of missing requirements?

Finally, after install python-memcache and reboot, log shows:

Connection OK
Traceback (most recent call last):
  File "/root/AlexaPi/main.py", line 144, in <module>
    token = gettoken()
  File "/root/AlexaPi/main.py", line 42, in gettoken
    refresh = refresh_token
NameError: global name 'refresh_token' is not defined

Actually not sure - other alexa projects seemed to require going to the browser and getting a token and putting it somewhere? Don't see instructions for that here.

braddo99, 2016-11-27:
After that last issue, checked the log and I can see that the auth_web.py module, when run, throws an error Traceback (most recent call last):
  File "./auth_web.py", line 5, in <module>
    from cherrypy.process import servers

Khemo2, 2016-12-04:
I have the same issue. Still have not gotten it to work.

braddo99, 2016-12-04:
Eventually I figured out that there's no such library as "easy_install" and pip also was not installing. I just went through and installed all of the requirements manually then the i stall co.pleted successfully. Once do e you can go to a web browser and browse to the referenced URL and I guess that causes the system to connect to Amazon.  Working now for me. Didn't get it going in time to get the Alexa deals for cyber Monday though, now I guess I can ask about the weather ;-)

Khemo2, 2016-12-04:
Thanks. I got it going. I have not done programming for nearly 40 years and I am a little rusty. Trying to do a little now that I have retired and have a little time.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment