alexa-pi/AlexaPiDEPRECATED #158: Raspberry Pi

jackofallsorts, 2017-01-05:
I am looking for Help
I have installed AlexaPi on top of the OSMC image.

When the OSMC boots, I hear the Hello, and if a MIC is not plugged in it keeps repeating the hello.

What I would like to do, is either enable a remote Bluetooth speaker which will act as Mic and Speaker for the AlexaPi

If this is not possible

Can someone guide me how to....or point me to a link that will show me how to....

separate the audio output only through the Stereo jack for the AlexaPi, and all output from the media center goes out through the HDMI.

Thank You Kindly

maso27, 2017-01-06:
Check out the newer repo at https://github.com/alexa-pi/AlexaPi
There's been a lot of development, including the ability to output media streams to a different device than Alexa's voice.

Basically, though, within config.yaml, you can select what input and output AlexaPi will use.  So that's a great place to start.

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment