alexa-pi/AlexaPiDEPRECATED #161: alsaaudio.ALSAAudioError: Device or resource busy [plughw:1]

ssendhil12, 2017-01-31:
Hi Guys,

I am using this module with Magic Mirror. Issue is first time I can hear alexa saying hello but after that I get the below error message.

alsaaudio.ALSAAudioError: Device or resource busy [plughw:1]

I am using a logitech c910 usb webcam with built in microphone. The raspberry pi 3 is connected to a TV via HDMI. These are the 2 connections.

I tried to search for plughw in the device variable in main.py based on other posting..Any pointers would be helpful.

(For some reason I am not able to make the usb sound to be the default as well).
Thanks

maso27, 2017-01-31:
@ssendhil12 not sure if you're using this fork or not, but you should be using [this one](https://github.com/alexa-pi/AlexaPi) instead.  Things have changed significantly.

Check the wiki, see if that covers your issues around audio.

ssendhil12, 2017-01-31:
@Maso27 ...thanks Yes I m using the same fork and I had gone thru the Wiki as well. Tried the stuff mentioned there. Still not had a resolution. Any comments pls let me know

maso27, 2017-01-31:
You'll probably get a lot more support if you post this issue in that repository instead of this one.

But here's a start:
1.)  You won't find _plughw_ in _main.py_.  It's located in _/etc/opt/AlexaPi/config.yaml_, where you can configure your audio inputs and outputs.
2.)  I'm not entirely certain if your _plughw_ error is an audio input or output, but since you're hearing the "hello", I'd guess it's the microphone.
3.)  Check `arecord -l` and/or `arecord -L` for hints about what device your microphone is, then you can play with what's in _config.yaml_ to see if the situation improves.  Realistically though I'd expect it to be _plughw:1_.
4.)  The dev branch at alexa-pi/AlexaPi has a _sox_ audio handler in addition to the _alsa_ one.  I haven't played with it but it might help clear things up too.

Are you running anything other than AlexaPi that might be listening for voice inputs?  Perhaps a different process is holding onto the microphone.

ssendhil12, 2017-01-31:
@maso27  - Thanks..Agreed I will comment here and post the issue in the repository.

1. Thanks Will take a look
2. Yes Im using a USB webcam (logitech c910) with microphone capability
3. Yes this shows the USB webcam
4. Ok thanks will check and keep you posted

ssendhil12, 2017-02-01:
@maso27 ..Thanks for the pointers...finally figured out that I had turned on the 'voicecontrol' module as part of the magic mirror. I disabled that...there is no more error. For some reason I am unable to awake 'Alexa' It says Hello and goes silent. No errors as well. arecord -l and -L shows the proper usb microphone. WIll keep checking...if u have any more pointers appreciate...Thanks.

ssendhil12, 2017-02-01:
finally made it to work..After having it working few times,..again getting the resource busy error...will check and keep you posted

ssendhil12, 2017-02-01:
finally it is working. Looks like I may need to get a proper usb microphone as it is not very effective. Thanks Maso27 for all the help.

closed 2017-02-01 by ssendhil12