alexa-pi/AlexaPiDEPRECATED #162: No sound from Raspberry Pi

Kinjalk, 2017-02-05:
I press the button - the red light comes on - then flashes 3 times - no sound comes out of the system.

Kinjalk, 2017-02-05:
Please help me. Also i am not getting anything in log

maso27, 2017-02-05:
Use the [newer version](github.com/alexa-pi/AlexaPi), follow the [wiki](https://github.com/alexa-pi/AlexaPi/wiki), then post any issues you might have there.  This version isn't really being developed at this point.

samiultahir, 2017-02-09:
Are your speakers loud enough an is your volume at a high level?

renekliment, 2017-03-01:
This project is now deprecated in favor of the new AlexaPi (https://github.com/alexa-pi/AlexaPi) which has all the features of this project and much more, such as:
- support for a wide range of devices and platforms (Raspberry Pi, CHIP, Orange Pi, desktops, Magic Mirror and others)
- voice activation (great PocketSphinx support and snowboy support soon to be released)
- much more robust installation and configuration
- better security
- Arch Linux support
- awesome documentation
- and much, much more! You get it - it's just awesome.

Please switch over to the new project and if your issue persists there, file an issue in the new repo's issue tracker. Thank you.

closed 2017-03-01 by renekliment